esp-rs/esp-hal #352: Add ability to do clock stretching

TheButlah, 2023-01-19:
I'm working with the bno080 imu and I suspect that I'm getting timeouts due to clock stretching.
How would I control clock stretching with the hal?

bjoernQ, 2023-01-19:
Currently we set the TO to 10 bus cycles - e.g. for C2, C3, S3 here https://github.com/esp-rs/esp-hal/blob/d29c4b1dcf682b3ac22e5381abd8f00ee559f462/esp-hal-common/src/i2c.rs#L584-L586

So, we should add a way to configure the TO (which is not yet possible). 

You might want to test if that really helps in your case by just changing it locally and see if that helps - if it does, we need to add the functionality to optionally pass the timeout

alexbohm, 2023-01-21:
I had a very similar issue with the BNO055 and locally increasing the timeout to 20 did solve the issue in my case.

TheButlah, 2023-01-22:
What is the maximum that I can configure the timeout to be that is supported by the hardware? Could the hal support something in software (but still use the hardware i2c as much as possible) if I need to go longer than that?

bjoernQ, 2023-01-23:
The max timeout is chip dependent but I think they can be configured to quite high values, e.g. on C3 _"The value of I2C_SCL_ST_TO_I2C should be less than or equal to 22, which means SCL_FSM could remain unchanged for 222 I2C_SCLK clock cycles at most before the interrupt is generated."_

I don't think there is a sane way to do it in a more software-controlled way

TheButlah, 2023-01-25:
esp-idf supports values quite large, at least as best as I can tell:
[image]
[image]

TheButlah, 2023-01-25:
> The max timeout is chip dependent but I think they can be configured to quite high values, e.g. on C3 _"The value of I2C_SCL_ST_TO_I2C should be less than or equal to 22, which means SCL_FSM could [...]
[image]

So yeah 2^22 clock cycles is *a lot*.

I did some testing locally on arduino and using the `Wire.setTimeOut()` function to increase the clock stretching limit does seem to be an important part of getting the bno080 to work. 

What are the next steps to add this functionality? I'm still probably not the best person to do this, but if no one has bandwidth I can try :)

Oooooh, `222` seemed quite low, but the datasheet actually says:

bjoernQ, 2023-01-25:
The `ticks_to_wait` doesn't look like it's the same thing: https://github.com/espressif/esp-idf/blob/49551cc48cb3cdd5563059028749616de313f0ec/components/driver/i2c.c#L1544-L1547

I think in my pasted text GH made the 2^22 into 222 - so yes 2^22 is a lot

We could add an optional parameter to `new` but that is a breaking change (not that much of a problem for the next release since that's a breaking release anyway - don't know when we will publish it but probably soon because of that `riscv` crate which got yanked) or add a second constructor which allows specifying the timeout. Then pass the value down to `set_fequency` which is very chip specific unfortunately (so we have a few of them).

I probably would prefer the second constructor but don't have a very strong opinion on that.

Would be awesome if someone who actually runs into that problem would implement it (e.g. I myself don't own any I2C device that needs it - it's certainly possible to use a second MCU to test it but might not be ideal)

TheButlah, 2023-01-25:
Part of the problem is that the clock stretching is only one factor in why the bno080 driver seems to fail in rust. But I could give it a try!

No promises though. And feel free if anyone sees this to implement it yourself, assume I'm gonna struggle bus for a while here. This would be my first real non-trivial contribution to the hal, I believe

TheButlah, 2023-01-30:
@bjoernQ I was looking into this. The `tout` value in [`esp-hal`](https://github.com/esp-rs/esp-hal/blob/c0eaa91a6f67b3b5f35652a3beae91bdc7f775f1/esp-hal-common/src/i2c.rs#L722) and the `i2c_set_timeout` function in [esp-idf](https://github.com/espressif/esp-idf/blob/236fa5e6698a262aed534d2a46c78711204b903a/components/hal/esp32s2/include/hal/i2c_ll.h#L239) both use the `I2C_TO_REG` (0x000C) *not* the `I2C_SCL_ST_TIME_OUT_REG` (0x0078).

[image]

Furthermore the datasheet shows that these two registers are doing slightly different things with either the SCL or SCL_FSM.
[image]

Im not sure which register is the correct one to use. What was the rationale for suggesting the use of the `I2C_SCL_ST_TIME_OUT_REG` instead of the one that the hal and esp-idf use? Should I switch from the the other register to using the one you suggested?

TheButlah, 2023-01-30:
So I investigated more. It looks like there are two separate timers. Each timer has a different interrupt, and a different range of values.

The first timer is controlled by `I2C_TIME_OUT_VALUE` (0x0C), and can range from [0, 2**5) APB clock
cycles. It is set [here](https://github.com/esp-rs/esp-hal/blob/c0eaa91a6f67b3b5f35652a3beae91bdc7f775f1/esp-hal-common/src/i2c.rs#L719-L725). This triggers the `I2C_TIME_OUT_INT` interrupt, which is [checked](https://github.com/esp-rs/esp-hal/blob/c0eaa91a6f67b3b5f35652a3beae91bdc7f775f1/esp-hal-common/src/i2c.rs#L946) to abort the i2c operation if it is fired. I see where we [disable](https://github.com/esp-rs/esp-hal/blob/c0eaa91a6f67b3b5f35652a3beae91bdc7f775f1/esp-hal-common/src/i2c.rs#L366-L374) the interrupt but couldn't find where we enable it though.

In `esp-idf`, the [`i2c_set_timeout()`](https://github.com/espressif/esp-idf/blob/8a7f6af6256f4c954053e08e819f55b93ddc1ff1/components/driver/i2c.c#L903) function calls [`i2c_ll_set_tout()`](https://github.com/espressif/esp-idf/blob/236fa5e6698a262aed534d2a46c78711204b903a/components/hal/esp32c3/include/hal/i2c_ll.h#L258-L269) to configure the timeout. The memory address is defined in [this struct](https://github.com/espressif/esp-idf/blob/236fa5e6698a262aed534d2a46c78711204b903a/components/soc/esp32c3/include/soc/i2c_struct.h#L75). Its the third union, so 3xu32 values before it means its the 12th==0x0C byte, which is indeed the address of `I2C_TIME_OUT_VALUE` relative to the base address of the register block.

What is interesting is that the comment on `i2c_ll_set_tout()` says that the `tout` parameter sets the timeout to `2**tout` APB clock cycles. That this contradicts what I discovered above - the [technical reference](https://www.espressif.com/sites/default/files/documentation/esp32-c3_technical_reference_manual_en.pdf#Regfloat.27.12) states that the time is not exponential in the value of the register. I.E. the max number of cycles is not `2**N` but rather just `N`. I'm going to assume that the technical reference is correct, and `esp-idf` should revise the comment.

The other counter that is useful but to my knowledge not used by either `esp-idf` or `esp-hal` is the on that @bjoernQ mentioned is [`I2C_SCL_ST_TO_I2C`](https://www.espressif.com/sites/default/files/documentation/esp32-c3_technical_reference_manual_en.pdf#Regfloat.27.9) (0x78) and can range from [0,23]. This will trigger an interrupt when `2**I2C_SCL_ST_TO_I2C` `I2C_SCLK` clock cycles have elapsed. Presumably, this has a less precise but far longer wall time than the other interrupt timer.

My questions are:
1. Which timer should we be using? I think we must choose between the high range (`I2C_SCL_ST_TO_I2C`) or high precision (`I2C_TIME_OUT_VALUE`) timers, right?
2. Does esp32 have those same two timers? Or does it only have the high precision one, but it has more bits to work with.
3. Is my understanding about the comment in `esp-idf` correct/should I open an issue there to have them confirm its wrong and have them change it?
4. Given this information about `tout` being non-exponential, does [this](https://github.com/esp-rs/esp-hal/blob/c0eaa91a6f67b3b5f35652a3beae91bdc7f775f1/esp-hal-common/src/i2c.rs#L584-L586) code still make sense? To me that should just be set to `10` since its not exponential.
5. With this in mind, does this behavior diverge at all from the esp32? Are both timer types available on all esp32 family devices?

bjoernQ, 2023-01-30:
Maybe I'm mixing up things but from what I see `I2C_TO_REG (0x000C)` is used in both, esp-hal and esp-idf

While in the TRM the register description doesn't really mention it's exponential it's mentioned in the chapter about the general timeouts.

In general, I'd say if TRM and esp-idf say contradicting things, usually esp-idf is right

From what I see on ESP32 there is only `I2C_TO_REG` - so that even more suggests we should continue to use that one

TheButlah, 2023-01-30:
> In general, I'd say if TRM and esp-idf say contradicting things, usually esp-idf is right

OK, I will submit feedback for the technical reference, as well as open an issue in esp-idf, one of those will hopefully get back to me and confirm that esp-idf is correct.

Do you have any suggestions on how I could verify that hypothesis? Like whats the best way for me to test the master device's clock stretching implementation? Is there a rust-friendly device that support slave clock stretching I could connect on a breadboard and program to stretch the clock a particular length, to test when exactly the master triggers a timeout?

bjoernQ, 2023-01-30:
I don't know of something like that but probably I can look into creating such a thing since it would also be helpful for smoke-testing of the HAL

bjoernQ, 2023-01-31:
> Is there a rust-friendly device that supports slave clock stretching I could connect on a breadboard and program to stretch the clock a particular length, to test when exactly the master triggers a timeout?

I hacked together something here: https://github.com/bjoernQ/i2c-dummy-device

It's not well written and definitely not well tested but hopefully be helpful

TheButlah, 2023-02-06:
Unfortunately I don't have bandwidth to work on this right now, but if I come back to it ill let y'all know

TheButlah, 2023-02-06:
BTW, I asked some unanswered questions in https://github.com/espressif/esp-idf/issues/10659. 

> usually esp-idf is right

In this case esp-idf says in [public documentation](https://github.com/espressif/esp-idf/blob/8a7f6af6256f4c954053e08e819f55b93ddc1ff1/components/driver/include/driver/i2c.h#L587) that the value is linear, this is also what the technical reference says.

In private documentation of the [low level code](https://github.com/espressif/esp-idf/blob/236fa5e6698a262aed534d2a46c78711204b903a/components/hal/esp32c3/include/hal/i2c_ll.h#L258-L269), it says that the value is exponential.

The esp-rs hal seems to treat the value as exponential (I think) because thats how esp32 works.

I would like some clarification on:
* Is the value linear or exponential?
* If its linear, should we switch to `I2C_SCL_ST_TO_I2C` on esp32c3, since that one *definitely* is exponential?

TheButlah, 2023-04-05:
Any clarification to be had here? I can try to implement this but I need more info.

bjoernQ, 2023-04-06:
What kind of clarification do you need? If the timeout is exponential?
I can't find a definite source of truth unfortunately but maybe it is possible to clarify things by experimenting

TheButlah, 2023-04-06:
the core question here that I can't find any information on is what I posted in https://github.com/espressif/esp-idf/issues/10659 two months ago. If I can at least get some clarification/understanding on that, it would be really nice.

bjoernQ, 2023-05-19:
There will be (soon) updated TRMs available with clarifications about the timeouts

M4tsuri, 2023-09-01:
I'm also working on BNO08x drivers and @alexbohm's solution works for me.

I also did an simple experiment to verify whether `I2C_TIME_OUT_VALUE` is exponential or linear with a BNO085.

**My assumptions**

1. `I2C_TIME_OUT_VALUE` uses the APB clock
2. BNO085 typically stretches line for about 400us  
   [image]
3. APB clock has a frequency of 80MHz

**Deduction**

- It takes about `0.0004 / (1 / 80_000_000) = 32000` clocks for an I2C timeout.
- `I2C_TIME_OUT_VALUE` is linear -> no enough bits to represent 32000 -> any value can lead to timeout
- `I2C_TIME_OUT_VALUE` is exponential -> setting it to at least `ceil(log2(32000)) = 15` is reasonable

So I modified the `tout` value in esp-hal-common/src/i2c.rs to 16 and the sensor behaves properly while a value of 14 makes the sensor timeout constently.

**Conclusion**

`I2C_TIME_OUT_VALUE` is exponential, i.e. TRM is wrong.

bjoernQ, 2023-09-07:
Thanks for analyzing this. I guess all we need to resolve this would be a function on I2C to set a custom timeout

bjoernQ, 2023-12-11:
With #1011 merged I think we can close this

closed 2023-12-11 by bjoernQ