esp-rs/esp-hal #358: Peripheral DMA - use peripheral dma status registers

MabezDev, 2023-01-23:
For example, DMA SPI we should be using `SPI_DMA_INT_ENA_REG` to track completion, _not_ the DMA interrupts, this is because although the DMA may have finished transferring the bytes, the peripheral might not have flushed them to the wire yet. This is why we also need a `spi.flush()` at the end of a transmission in the current implementation.

bjoernQ, 2023-01-24:
Yes, unfortunately waiting just for the DMA transfer to pushed all data to a peripheral is not enough since the DMA controller doesn't know about the buffers of a peripheral.

Apparently not all DMA-capable peripherals have dedicated DMA interrupts - e.g. for I2S we need to rely on the idle interrupt to know the transfer completed.

In regards of SPI, I'm not sure if it makes much difference if we wait for the operation to complete or if we use the DMA interrupt - in my tests they seem to be interchangeable but maybe your use-case is different.

In any case it's probably saner to use the DMA interrupt when dealing with DMA.

MabezDev, 2023-01-24:
> Apparently not all DMA-capable peripherals have dedicated DMA interrupts - e.g. for I2S we need to rely on the idle interrupt to know the transfer is completed.

True, but I suppose the point is we know to register the peripherals interrupt handler, not the DMA specific one - which we will need to do for the esp32/esp32s2 anyway.

> In regards of SPI, I'm not sure if it makes much difference if we wait for the operation to complete or if we use the DMA interrupt - in my tests they seem to be interchangeable but maybe your use-case is different.
>
> In any case it's probably saner to use the DMA interrupt when dealing with DMA.

This is true, the FIFO is only 64bytes so even at slow speeds like `100.khz()` it will only block on flush for `0.00064s`, so not really a big deal, even for async applications.

I guess for blocking/interrupt modes it doesn't matter what we do here because end users need to setup the interrupt handlers. For async we can easily start the transfer and wait on the peripheral-specific interrupt for each driver where appropriate. So I guess there is nothing to do here.

closed 2023-01-24 by MabezDev