esp-rs/esp-hal #361: Implement `async` support for all applicable peripheral drivers

jessebraham, 2023-01-25:
While we have a tracking issue for async drivers _with_ `embedded-hal-alpha` traits (#70), we should additionally implement async drivers for peripherals which _do not_ have traits pre-defined. We can still provide our own APIs and make these useful.

As we make progress on #70 and I have a better idea of how things are looking, I will update this comment to (eventually) include a task list tracking which drivers need work.

playfulFence, 2023-06-19:
I'll take a look at `usb-serial-jtag` as a start!

JurajSadel, 2023-06-19:
I'll look into `AES`

SergioGasquez, 2023-06-22:
I'll be working on SHA!

jessebraham, 2023-07-24:
Async support for USB Serial JTAG added in #632

jessebraham, 2023-08-22:
Maybe it would be helpful to go through our existing drivers and figure out which are capable of asynchronous operation, then we can create a task list here to track the progress and hopefully make it easier for people to contribute.

playfulFence, 2023-09-05:
I'm working on `RSA`

jessebraham, 2023-09-05:
@JurajSadel @SergioGasquez any updates on your progress? If you don't have time to work on these it might be nice to push your current efforts to a branch so somebody else could wrap it up.

SergioGasquez, 2023-09-05:
I started working on SHA, abandoned it (it was not working) for other stuff, then DMA was refactored in several ways. Today, I started reimplementing my changes from scratch using an updated base branch, here are my changes: https://github.com/esp-rs/esp-hal/compare/main...SergioGasquez:esp-hal:feat/dma-sha. It still not working, so I will try to work on it

JurajSadel, 2023-09-07:
@jessebraham I have my work here: https://github.com/JurajSadel/esp-hal/tree/feature/async-aes it does something but it's not correct yet.

bjoernQ, 2023-09-18:
I will start looking into async I2S next

bjoernQ, 2023-09-21:
I'll implement async PARL_IO next

jaedson-barbosa, 2023-09-29:
Hi, the async read i2c method in esp32c3 is not working. 😄

playfulFence, 2023-11-16:
Will take a look into `USB` for `s2` and `s3`

ptpaterson, 2023-11-17:
I'm working on `TWAI` in #951

jessebraham, 2024-02-28:
Just noting, the list in the original comment needs to be updated to include new peripherals on the ESP32-P4 (but this is pretty low-priority).

jessebraham, 2024-10-03:
I think this issue has probably outlived its usefulness, we can create issues per-driver as needed (some of these exist already).

closed 2024-10-03 by jessebraham