esp-rs/esp-hal #367: Linker error with latest `esp32c3-hal` release

jessebraham, 2023-01-28:
Xtensa HALs don't seem to be affected. Didn't test the C2 yet because `esp-template` hasn't been updated to include it yet and I'm lazy, but don't imagine it's affected; will verify during the work week.

```text
  = note: rust-lld: error: undefined symbol: ets_update_cpu_frequency
          >>> referenced by esp_hal_common.8c1a008e-cgu.13
          >>>               esp_hal_common-fb1c4a86c92af4e5.esp_hal_common.8c1a008e-cgu.13.rcgu.o:(esp_hal_common::clock::clocks_ll::esp32c3_rtc_update_to_xtal::h7df7af85b4d084d9) in archive /Users/jesse/Sandbox/sucrose-vm/target/riscv32imac-unknown-none-elf/release/deps/libesp_hal_common-fb1c4a86c92af4e5.rlib
          >>> referenced by esp_hal_common.8c1a008e-cgu.13
          >>>               esp_hal_common-fb1c4a86c92af4e5.esp_hal_common.8c1a008e-cgu.13.rcgu.o:(esp_hal_common::clock::clocks_ll::esp32c3_rtc_freq_to_pll_mhz::h2d2bfe39e10ca900) in archive /Users/jesse/Sandbox/sucrose-vm/target/riscv32imac-unknown-none-elf/release/deps/libesp_hal_common-fb1c4a86c92af4e5.rlib
```

Not sure how this built locally and in CI up until the point of release for all of us, but I guess we'll have to fix it and publish a patch release.

bjoernQ, 2023-01-30:
I just tried generating a new project for ESP32-C3 and it compiles fine for me

Also `esp-template's`'s CI seems to be fine: https://github.com/esp-rs/esp-template/actions/runs/4041707622/jobs/6948553337

jessebraham, 2023-01-30:
It works on my Macbook (Ventura, x86_64) but not on my Mac Mini (Ventura, aarch64), and I've seen at least one other person mention they saw this error. Not sure what the difference between my dev environments is, tried for about 15 minutes to get it building on my Mac Mini.

ImUrX, 2023-01-31:
I just tried it locally and it builds. But CI doesn't build and I just invalidated the whole cache in case it was that but it still happens. [Logs](https://github.com/SlimeVR/SlimeVR-Rust/actions/runs/4051710355/jobs/6970630633)

bjoernQ, 2023-01-31:
Ok I see why it's not compiling now:
- https://github.com/esp-rs/esp-hal/blob/main/esp32c3-hal/ld/rom-functions.x#L9 defines `ets_update_cpu_frequency_rom`
- https://github.com/esp-rs/esp-hal/blob/main/esp-hal-common/src/clock/clocks_ll/esp32c3.rs#L11 wamts `ets_update_cpu_frequency`

So, when using `ClockControl::boot_defaults` it should work since that doesn't call that function but `ClockControl::configure` needs that function

_BUT_ it's defined like this for all targets - not just ESP32-C3 🤷‍♂️ (So it should fail for all targets)
Also seeing it build in some environment while it fails in others makes no sense at all

jessebraham, 2023-01-31:
I guess we should maybe have at least one example which does _not_ use the boot defaults in order to exercise this in CI. Regardless, will have a PR fixing the references to that function shortly.

closed 2023-01-31 by jessebraham

ImUrX, 2023-02-03:
@jessebraham Can this be backported to 0.5.1 or something like that?

TheButlah, 2023-02-13:
> @jessebraham Can this be backported to 0.5.1 or something like that?

now that 0.6 is coming out soon, we will just wait for that

PR #374 (2023-01-31): Fix the references to the `ets_update_cpu_frequency_rom` ROM function
Closes #367 (I think 😁). It was already working on this system to begin with, I'll hop on my other computer in a bit and test it out there.