esp-rs/esp-hal #368: Panics on spi.write() in debug build

averyanalex, 2023-01-29:
App panics when [pn532](https://github.com/WMT-GmbH/pn532) library (`pn532 = { git = "https://github.com/WMT-GmbH/pn532.git", rev = "ddf192df9f3da5c5b220b3be66593a2c206127b9", features = ["msb-spi",] }` tries to send data to PN532 module in **debug** build (in release build everything works as expected).
Board: esp32c3.
Boot mode: tested esp-idf bootloader and direct-boot.
Esp-hal version: 0.5.0.
If matters, I use riscv-rt instead of esp-riscv-rt because of linker errors.
Panic log:
```
!! A panic occured in '/nix/store/dwv7swj2j78ap6k0l3fz215k2k66sjbs-rust-default-1.69.0-nightly-2023-01-29/lib/rustlib/src/rust/library/core/src/panicking.rs', at line 123, column 5

PanicInfo {
    payload: Any { .. },
    message: Some(
        unsafe precondition(s) violated: ptr::copy_nonoverlapping requires that both pointer arguments are aligned and non-null and the specified memory ranges do not overlap,
    ),
    location: Location {
        file: "/nix/store/dwv7swj2j78ap6k0l3fz215k2k66sjbs-rust-default-1.69.0-nightly-2023-01-29/lib/rustlib/src/rust/library/core/src/panicking.rs",
        line: 123,
        col: 5,
    },
    can_unwind: false,
}

Backtrace:

0x4204c8ca
0x4204c8ca - core::panicking::panic_nounwind
    at /nix/store/dwv7swj2j78ap6k0l3fz215k2k66sjbs-rust-default-1.69.0-nightly-2023-01-29/lib/rustlib/src/rust/library/core/src/panicking.rs:123
0x4201e212
0x4201e212 - core::intrinsics::copy_nonoverlapping::runtime
    at /nix/store/dwv7swj2j78ap6k0l3fz215k2k66sjbs-rust-default-1.69.0-nightly-2023-01-29/lib/rustlib/src/rust/library/core/src/intrinsics.rs:2223
0x4200c8d6
0x4200c8d6 - esp_hal_common::spi::Instance::write_bytes
    at /home/alex/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-hal-common-0.5.0/src/spi.rs:1410
0x420025e8
0x420025e8 - <esp_hal_common::spi::Spi<T> as embedded_hal::blocking::spi::Write<u8>>::write
    at /home/alex/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-hal-common-0.5.0/src/spi.rs:258
0x42014d0e
0x42014d0e - <pn532::spi::SPIInterface<SPI,CS> as pn532::Interface>::write
    at /home/alex/.cargo/git/checkouts/pn532-5e2e68573209ed62/ddf192d/src/spi.rs:64
0x42008906
0x42008906 - pn532::protocol::Pn532<I,T,_>::_send
    at /home/alex/.cargo/git/checkouts/pn532-5e2e68573209ed62/ddf192d/src/protocol.rs:201
0x42008b62
0x42008b62 - pn532::protocol::Pn532<I,T,_>::_process
    at /home/alex/.cargo/git/checkouts/pn532-5e2e68573209ed62/ddf192d/src/protocol.rs:102
0x420089e8
0x420089e8 - pn532::protocol::Pn532<I,T,_>::process
    at /home/alex/.cargo/git/checkouts/pn532-5e2e68573209ed62/ddf192d/src/protocol.rs:93
0x4200e808
0x4200e808 - main
    at /home/alex/projects/averyanalex/multikey/src/main.rs:121
0x42000104
0x42000104 - _start_rust
    at /home/alex/.cargo/registry/src/github.com-1ecc6299db9ec823/riscv-rt-0.11.0/src/lib.rs:424
```

bjoernQ, 2023-01-29:
Thanks  for reporting this 
This is most probably the same problem as #295

averyanalex, 2023-01-31:
> Thanks for reporting this This is most probably the same problem as #295

Hm, I think yes, it is duplicate, sorry)

bjoernQ, 2023-02-01:
Are you fine with closing this as a duplicate? Re-open if you disagree

closed 2023-02-01 by bjoernQ