esp-rs/esp-hal #375: Implement support for light-/deep-sleep

jessebraham, 2023-02-01:
This is mostly just a placeholder for now, I will update this text when I have more information to share on the topic.

jessebraham, 2023-07-14:
Support for light-/deep-sleep for ESP32 added in #574

bugadani, 2023-07-18:
I'd like to add as a todo item that the HAL should probably not expose the `RtcSleepConfig` type(s) directly. Instead, the HAL should expose something akin to esp-idf's "sleep flags". The reason is, that currently it is possible to create subtly broken sleep configurations: the user is free to set whatever combination they want, and the driver has no way of verifying the sleep configuration. Wakeup sources may also need to take care of setting up correct values, which should be easier handled when applying sleep flags.

Ben-PH, 2023-08-09:
I'd like to contribute to this. Have an important use-case for C3, and a nice-to-have use case with S3 for light-sleep.

I have no idea how to go about doing this. I'll read through the links, but embedded and EE stuff is a bit crazy to me.

This will be my first time contributing to this project, and couldn't find a contributing file/instruction documentation. How should I go about contributing? fork-draft PR-iterate?

@jessebraham should we coordinate on this one?

jessebraham, 2023-09-29:
ESP32-C3 support add in #795

bugadani, 2023-11-09:
I'm working on C6 now. It's a bit annoying as function names are different, but the flow seems to be similar enough. May take some time but I'll hope to get it working in a week or two.

alevinval, 2024-06-17:
Hi folks! I wanted to ask a question. Is there a particular reason for those delays introduced before sleeping? See here https://github.com/esp-rs/esp-hal/blob/main/examples/src/bin/sleep_timer.rs#L37-L40

I do realise that my esp32s3 sometimes never wakes-up after entering deep sleep with a timer wake-up source. I am not sure if the cause is that I do not delay enough, or that I use `--release` mode which optimizes things away. So... is there a good reason to delay before entering deep sleep? And how big should that delay be relative to the clock?

closed 2026-01-14 by MabezDev