esp-rs/esp-hal #377: Add support for the RISC-V ULP

jessebraham, 2023-02-01:
This is mostly just a placeholder for now, I will update this text when I have more information to share on the topic.

jessebraham, 2023-03-29:
I have begun work on the lower-level aspects of adding support for the ESP32-S2/S3: I have created initial versions of their SVDs and am able to generate PACs from these.

The next step is to load a basic application to the RISC-V ULP core and execute it. I unfortunately have not yet been successful with this, but hopefully once we start working on this project again we can get to this point without too much hassle.

jessebraham, 2023-07-10:
SVDs and PACs have been created for the ESP32-C6's LP core and the ESP32-S2/S3's ULP cores.

bjoernQ, 2023-07-20:
Initial work on ESP32-C6 LP core is started in https://github.com/bjoernQ/esp32c6lp-hal - as soon as we are confident enough it's heading into the right direction the code will get moved to the esp-hal repo

jessebraham, 2023-08-14:
#714, #715, and #720 have begun work on the `esp32c6-lp-hal` package in this repository. More features to come.

bjoernQ, 2023-10-04:
Started working on S2/S3 ULP HAL. Initial version might not support any peripherals (depends on the progress I make) but will enable us to compile Rust code to run on the ULP

jessebraham, 2023-11-14:
I think we can probably close this, we have `esp32c6-lp-hal` and `esp-ulp-riscv-hal` packages now, and while not feature complete they are present and functional; enough to build on top of, at least. I think it would be better if we opened more, finer-grained issues instead.

So, gonna close this. If anybody has any objections we can re-open the issue.

closed 2023-11-14 by jessebraham

maufl, 2024-07-23:
Is there an minimal example on how to use these crates? I have successfully written a small Rust program for my ESP32-C6 and now I'd like to include a program for the (U)LP co-processor. The `exp32c6-lp-hal` crate has examples, but I don't know how to build them, include them when flashing and how to load them from my main program.

MabezDev, 2024-07-23:
We have examples using the LP core in the top level examples folder: https://github.com/esp-rs/esp-hal/blob/5d6354ccbd62527452968074c4a92a4fc6c6a635/examples/src/bin/lp_core_basic.rs#L33-L43

maufl, 2024-07-23:
Perfect! This is exactly what I was looking for, and such a fast response time :star_struck: Thank you very much :pray:

jessebraham, 2024-07-23:
I do also plan on spending some time further developing the low-power HAL in the near future. Probably not this release cycle, but possibly the next. This will include more/better documentation as well, I hope to make the process a bit easier.