esp-rs/esp-hal #381: Type erase SpiDma once created

MabezDev, 2023-02-03:
It is currently impossible to declare SpiDma as a type because of its use of private types.

```rust
pub type SpiType<'d> = SpiDma<'d, esp32c3_hal::peripherals::SPI2, esp32c3_hal::dma::private::ChannelTx<'d, gdma::private::Channel0TxImpl, gdma::private::Channel0>, esp32c3_hal::dma::private::ChannelRx<'d, gdma::private::Channel0RxImpl, gdma::private::Channel0>, gdma::private::SuitablePeripheral0>;
```

The easy option is to make these modules public and then seal the traits etc, but it would be better in general if we could completely erase this type information _once_ we've satisfied the trait bounds. Any ideas on how to go about this @bjoernQ?

This is somewhat blocking async SPI as its impossible to pass the spi struct into a embassy task.

bjoernQ, 2023-02-03:
Having a way to erase the types would be very convenient - probably also for other use-cases - but sounds like a lot of work.
I guess to unblock your work on async-SPI it's fine to go the easy way for now and later we can still add a way to erase the types

bugadani, 2023-07-05:
"Closes" directives don't work in comments, so you guys should probably close this :)

closed 2023-07-05 by jessebraham