esp-rs/esp-hal #386: Embassy examples don't work with direct-boot

MabezDev, 2023-02-07:
Investigating now.

MabezDev, 2023-02-07:
It seems this is nothing to do with embassy, but the usage of embassy triggers the issue. It seems taht interrupt handlers above 13 have there handlers trashed in memory with garbage.

MabezDev, 2023-02-07:
Things I've ruled out so far:

- A bug in moving to esp-riscv-rt - it still fails before we moved.
- stack, I increased the hart stack to 32K (hopefully this value is actually reflected).
- dodgy atomic emulation - I removed the atomic emulation cfgs and rebuilt, the error persists.

bjoernQ, 2023-02-17:
It seems to me that the interrupt gets dispatched fine but the target in RAM is just zero-bytes

```
Init!
Waiting...
Ping!
Exception 'Illegal instruction' mepc=0x40380932, mtval=0x00000000
0x40380932 - interrupt15
    at C:\projects\esp\esp-hal\esp-hal-common\src\interrupt\riscv.rs:417
```

This is the disassembly
```
40380932 <interrupt15>:
    pub unsafe fn interrupt15(context: &mut TrapFrame) {
40380932:	85aa                	mv	a1,a0
        handle_interrupts(CpuInterrupt::Interrupt15, context)
40380934:	453d                	li	a0,15
40380936:	fffff317          	auipc	t1,0xfffff
4038093a:	71a30067          	jr	1818(t1) # 40380050 <_ZN14esp_hal_common9interrupt8vectored17handle_interrupts17hf2ed2d5a1dc67c82E>
```

And this is what I read from those addresses
```
0x40380932 = 00000000
0x40380933 = 00000000
0x40380934 = 00000000
0x40380935 = 00000000
0x40380936 = 00000000
0x40380937 = 00000000
0x40380938 = 00000000
0x40380939 = 00000000
0x4038093a = 00000000
0x4038093b = 00000000
0x4038093c = 00000000
0x4038093d = 00000000
0x4038093e = 00000000
0x4038093f = 00000000
0x40380940 = 00000000
0x40380941 = 00000000
```

Not placing `interrupt15` in RAM makes the example work for me

So, it seems to me like this doesn't work as intended: https://github.com/esp-rs/esp-hal/blob/9fa59fe12754b367cb0b75668dbd896e8a78fba8/esp32c3-hal/src/lib.rs#L129-L142

bjoernQ, 2023-02-17:
Seems like  the addresses of _srwtext, _erwtext and _irwtext make sense - however manually copying the data to ram gives another immediate crash.

PR #396 (2023-02-20): Fix direct boot linker script for ESP32-C2/C3
Fixes #386 

This makes sure RWTEXT isn't overwritten by BSS
closed 2023-02-20 by MabezDev