esp-rs/esp-hal #403: Add support for the ESP-8266

Tastaturtaste, 2023-02-24:
The currently available [esp8266-hal](https://crates.io/crates/esp8266-hal) is in maintenance mode with no indication about its state. Since this seems to be the place for all the other esp-hals given the repositories name, would it be possible to integrate the crate into this repository and, if necessary, continue development?

jessebraham, 2023-02-24:
We will not be supporting the ESP8266 in this repository. There are no peripherals in this device which will share drivers with any of the ESP32* devices, and as such it does not make sense to include in this repository.

I and others have tried repeatedly to find anybody to maintain `esp8266-hal` over the years, and while plenty of people want to use it nobody is willing to put any work into it.

closed 2023-02-24 by jessebraham

Tastaturtaste, 2023-02-24:
Understandable, thank you for your answer and your work in this area.