esp-rs/esp-hal #438: ESP32-C2: Examples reset by super-wdt in direct-boot mode

bjoernQ, 2023-03-14:
e.g. `cargo build --example systimer --features direct-boot && espflash flash --format=direct-boot --monitor target\riscv32imc-unknown-none-elf\debug\examples\systimer`

Results in
```
ESP-ROM:esp8684-api2-20220127
Build:Jan 27 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SYSTIMER Current value = 626782
Interrupt lvl1 (alarm0)
Interrupt lvl1 (alarm0)
Interrupt lvl2 (alarm1)
Interrupt lvl1 (alarm0)
Interrupt lvl2 (alarm2)
ESP-ROM:esp8684-api2-20220127
Build:Jan 27 2022
rst:0x12 (SUPER_WDT_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
SYSTIMER Current value = 627306
Interrupt lvl1 (alarm0)
Interrupt lvl1 (alarm0)
Interrupt lvl2 (alarm1)
...
```

MabezDev, 2023-03-14:
This will most likely be the case for C3/C6 too? AFAIK we don't disable super wdt in esp-hal. The esp-idf bootloader sets it to auto-feed, so using direct boot will miss this step.

bjoernQ, 2023-03-14:
Just had a look: Some examples contain `rtc.swd.disable();` and some not - and it's different for different targets. So probably we should just add it for all the examples - or to none of them ... this way it's at least very confusing behavior without looking into the code