esp-rs/esp-hal #455: Is PSRAM supported?

teotwaki, 2023-03-26:
Hi,

I was wondering if there was any support for PSRAM, and if so, if anyone would have an example on how to use it. I'm already using the global allocator, however I believe I only have access to the 300k of on-board RAM.

Thanks!

bjoernQ, 2023-03-27:
Short answer: it's not there quite yet

Previously (in the first attempt on a HAL for ESP32) we relied on the bootloader to setup PSRAM but here we need to implement those things on our side

Secondly, we need to teach `esp-alloc` about the additional RAM to use it as heap memory.

bjoernQ, 2023-03-27:
A good start might be to port e.g. https://github.com/apache/nuttx/blob/master/arch/xtensa/src/esp32/esp32_spiram.c , starting with ` esp_spiram_init` which might require more code ported (like https://github.com/apache/nuttx/blob/master/arch/xtensa/src/esp32/esp32_psram.c )

And then do the same for ESP32-S2 and ESP32-S3.

Depending on which code base you are more familiar with, looking at this functionality in esp-idf might or might not be a better option.

When all that works PSRAM should be accessible via usual read and writes to the memory region.

If that is done adapting `esp-alloc` to always allocate in PSRAM should be trivial - having a way to allocate from both internal RAM and PSRAM at the same time via `esp-alloc` seems tricky currently and might be something to figure out later.

I'm not sure if this should be noted somewhere in the documentation currently since in general, we document what is support - not what is not yet supported. Also, with this we now have an issue for it

bjoernQ, 2023-04-26:
https://github.com/esp-rs/esp-hal/pull/493 added basic PSRAM support for ESP32-S2

Support for other targets will follow

closed 2023-05-11 by bjoernQ