esp-rs/esp-hal #469: SPI slave support

LinusU, 2023-04-05:
(first of all, thanks for all the amazing progress with Rust support for the ESP32, it truly is amazing!)

I'm trying to build an ISP slave device using the ESP32, and of course I want to build it using Rust. However, it _seems_ like there is no support fir this currently?

If I understand correctly, the SPI traits in embedded-hal is only for SPI master, and when slave traits will be added they will be separate and probably named spi_slave. ref: https://github.com/rust-embedded/embedded-hal/pull/396. And as far as I understand, esp-hal is an implementation of embedded-hal?

It seems like the only way right now is to use Rust with the ESP IDF (e.g. using ), and then call the C functions (`spi_slave_initialize`, `spi_slave_transmit`, etc.) via the `esp_idf_sys` crate?

If my understanding is correct, than this is a feature request to add SPI slave support. In either case, any pointers or guidance would be much appreciated!

bjoernQ, 2023-04-06:
Yes, we implement SPI traits from embedded-hal which all define only 1 bit full-duplex master mode.
We also have some additional functionality like half-duplex, dual and quad SPI

Technically we can define our own API to add SPI slave support (like we had to do for e.g. I2S before)

TL;DR this is a feature request for SPI slave functionality 😄

Studiedlist, 2023-10-13:
As far as I understand from reading #580, slave driver for ESP32 and ESP32-S2 is not implemented due to `pdma` issues. 

I could try to investigate this and add implementation if someone guide me through codebase a little bit so I can understand how to get started

Studiedlist, 2023-10-14:
I browsed through available spi options and found some things that could help make it work. 
According to `esp32::spi0::clock::W` documentation and esp32 datasheet, all of its options should be set to 0 in slave mode.
So I added the following to `start_read_bytes_dma`:

```rust
reg_block.clock.modify(|_, w| {
    w.clkdiv_pre().variant(0)
    .clk_equ_sysclk().clear_bit()
    .clkcnt_l().variant(0)
    .clkcnt_h().variant(0)
    .clkcnt_n().variant(0)
});
```
There are also `dma_int_ena::in_done_int_ena` and `dma_int_ena::in_suc_eof_int_ena` bits that enable dma interrupts, but I don't know are they used in current driver. 

The next observation is that  `slave::slv_last_state_bit` is working correct, so, I think, dma is handling transaction, but there is something with `ch.tx` and `ch.rx`. 

Unfortunately, I cannot find actual `Rx` and `Tx` implementation since `Spi2DmaChannelRxImpl` is generated with macro, so I cannot debug rx/tx internals

Studiedlist, 2023-10-14:
I have also written sketchy interrupt-based no-cs slave driver:
```rust
#[interrupt]
fn GPIO() {
    critical_section::with(|cs| {
        let mut slave = CLK.borrow_ref_mut(cs);
        let slave = slave.as_mut().unwrap();

        let mut timer = TIMER0.borrow_ref_mut(cs);
        let timer = timer.as_mut().unwrap();

        timer.start(100u64.millis());

        let input = slave.mosi.is_high().unwrap();

        slave.end = false;

        slave.buf_n += (input as u8) << (7 - slave.c);
        slave.c += 1;
        if slave.c == 8 {
            slave.c = 0;
            slave.buf.push(slave.buf_n);
            slave.buf_n = 0;
        }
        slave.clk.clear_interrupt();
    });
}

#[interrupt]
fn TG0_T0_LEVEL() {
    critical_section::with(|cs| {
        let mut timer = TIMER0.borrow_ref_mut(cs);
        let timer = timer.as_mut().unwrap();

        let mut slave = CLK.borrow_ref_mut(cs);
        let slave = slave.as_mut().unwrap();

        if slave.end == false {    
            println!("{:?}", slave.buf);
            slave.buf = Vec::with_capacity(110);
            slave.c = 0;
            slave.buf_n = 0;

        }
        slave.end = true;

        if timer.is_interrupt_set() {
            timer.clear_interrupt();
        }
    });
}
```
It works fine except low speed (about 10kHz). 
Could such approach be correct for no-dma implementation?

bjoernQ, 2023-10-17:
Thanks for working on this!

Regarding the interrupt-based implementation I guess we shouldn't include bit-banging protocol implementations in the HAL. Those could go into a 3rd party crate I guess

Studiedlist, 2023-12-09:
Unfortunately, I was unable to figure out what was wrong with ESP32, but I've got ESP32S3, and now test SPI slave with it.
[Example](https://github.com/esp-rs/esp-hal/blob/main/esp32s3-hal/examples/spi_slave_dma.rs) is working with Raspberry PI as master, but whenever I change `dma_transfer` to `dma_read`, `wait` call is stuck forever. 

I browsed through source code, and spotted [strange check](https://github.com/esp-rs/esp-hal/blob/main/esp-hal-common/src/spi/slave.rs#L347C6-L347C47) at `DmaTransfer::is_done` for `dma_read`. I've removed `ch.tx.is_done()`, and now `dma_read` is working. 
Could it be a mistake, or `ch.tx.is_done()` is required in case of `dma_read`?

Studiedlist, 2023-12-09:
By the way, current implementation of SPI slave for ESP32S3 is capable of handling insane clock speed of 10mHz.  
Thank you all for this feature

bjoernQ, 2023-12-11:
> Unfortunately, I was unable to figure out what was wrong with ESP32, but I've got ESP32S3, and now test SPI slave with it. [...]

I think the intent is to make sure that all bits are clocked out on MOSI - but probably a good hint what to look at. Maybe just checking `rx.is_done()` should be good enough in reality

Studiedlist, 2023-12-11:
Now I see the reason: there is only one implementation of `wait` for both read and write transfers, so both `ch.tx.is_done()` and `ch.rx.is_done()` are required. 

According to datasheet, `GDMA_OUT_TOTAL_EOF_CH` interrupt, which is used to determine if tx channel is done:
> triggered when alll data corresponding to a linked list (including
multiple descriptors) have been sent via transmit channel n

There is no exact information on whether it is fired when tx list is empty. But, as far as I understand, no transmission takes place when there is nothing to send, so `GDMA_OUT_TOTAL_EOF_CH` will not be triggered.

I think, the problem can be solved by introducing different `SpiDmaTransfer` structs for read and write, like `SpiDmaTransferRead` and `SpiDmaTransferWrite`.

jessebraham, 2024-04-11:
We're relatively close to the finish line here I think, so let's try to get this issue closed in the next few weeks/months if able :) I will mention this issue during our team meeting on Monday.

JurajSadel, 2024-05-17:
Update: PR with S2 support https://github.com/esp-rs/esp-hal/pull/1562. I'm not sure about ESP32, we've spent long time on that but the thing doesn't work.

closed 2024-10-07 by JurajSadel