esp-rs/esp-hal #483: Adding priority based interrupt preemption

onsdagens, 2023-04-19:
Hi! I've been working on porting the RTIC framework to the ESP32-C3. It relies heavily on interrupt priorities and preemption for hardware acceleration of scheduling. So far i've had it implemented pretty messily in a fork, but now i have some time to make it look nice and possibly get the feature merged. I've been moving stuff around for the past day, and i'm pretty happy with how it looks at this point.

My proposed changes are available here:

https://github.com/onsdagens/esp-hal/tree/interrupt_preemption

, i've also added an example of using the feature raw (outside of rtic) in the esp32-c3 hal. For now it's the esp32-c3 only since it's the chip i have available, but i think that this is definitely possible to implement on other esp32 chips that have a priority threshold register for interrupt filtering.

I'd love to hear your take on this!

onsdagens, 2023-04-19:
I've also been able to move all of the necessary changes to the hal, meaning we don't have to touch the RT at all, which is nice i think.

sethp, 2023-05-01:
Oh, this looks neat! We ended up doing something similar on a handler-by-handler basis using a closure: [`fn interrupt_mask(..)`](https://github.com/rustbox/esp32c3-vgaterm/blob/7d5a88599a9c70dad54c9b5975ef2de07d2b26d9/src/interrupt.rs#L117) ([example usage](https://github.com/rustbox/esp32c3-vgaterm/blob/7d5a88599a9c70dad54c9b5975ef2de07d2b26d9/src/uart.rs#L218)), but I like the idea of a global opt-in. We also considered adding a "nestable" attribute to the `#[interrupt]` macro, which would more or less inject the code you've got here into the trampoline around the call to the user-specific function.

I've got a couple notes for your specific implementation, but I think it'll be easier to discuss those over on #448.

sethp, 2023-05-01:
Oh, I guess that's actually a bit different of an implementation, isn't it? My bad!

Besides the two comments I left over there, I'd also suggest either tagging those functions with `#[inline(always)]`, or equivalently, turning them into inline blocks in the caller like so:

```rust
pub unsafe extern "C" fn start_trap_rust_hal(trap_frame: *mut TrapFrame) {
// ...
        #[cfg(feature = "interrupt-preemption")]
        let interrupt_priority = {
             let interrupt_id: usize = mcause::read().code(); // MSB is whether its exception or interrupt.
             // ...
             prev_interrupt_priority
        };
```

Trap service entry is, at least for us, extremely latency sensitive; I think this is fairly common though? I would expect it toimpact your RTIC framework fairly heavily, if you're doing a lot of interrupt-servicing. The problem with allowing LLVM to choose when to inline that function is that it's not aware of the huge performance (and variability) cliff when a function moves from IRAM to flash, so it's not making a decision based on the full set of costs.

An alternative would be to tag them both with `#[link_section = ".trap.rust"]` so they'll end up going wherever `_start_trap_rust` goes, which at least helps re-aligns the cost model LLVM "sees" with the machine. That said, it seems strictly worse to me to not inline in this case, which is why I'd lean toward an approach that models that explicitly.

Sorry for the fragmented feedback!

onsdagens, 2023-05-03:
> Oh, this looks neat! We ended up doing something similar on a handler-by-handler basis using a closure: [`fn [...]

RTIC is based on generating interrupt handlers via macros, so i ended up adding this functionality directly to the framework via those macros to avoid pulling in forks. 

Nevertheless, i agree with you on all points, i'm working on fixing the errors you've mentioned and merging the branch mentioned in the issue with the PR branch to avoid confusion. Thanks for the feedback!

onsdagens, 2023-05-03:
I've implemented the suggestions now, added some examples for the other RISC-V as @bjoernQ mentioned, i can't test those however since i don't own them.

sethp, 2023-05-03:
Wow, great work! Thanks for the quick turnaround, and I'm glad you found my suggestions helpful.

closed 2023-05-05 by onsdagens

gyzerok, 2023-06-11:
Hello @onsdagens!

I've tried to search across this repo, but can't find any examples of using ESP32 C3 with RTIC, that you mentioned. Can you please point me in the right direction?

Thank you for working on it!

onsdagens, 2023-06-11:
> Hello @onsdagens!
> 
> 
> 
> I've tried to search across this repo, but can't find any examples of using ESP32 C3 with RTIC, that you mentioned. Can you please point me in the right direction?
> 
> 
> 
> Thank you for working on it!

Hi! Here are some sanity check examples I've thrown together: https://github.com/perlindgren/esp32c3-test/tree/rtic_sw_tasks
, this along with the Cortex-M examples provided with RTIC itself should be enough to get you started. I'm currently on vacation but once I get back I'll be looking into finally getting the port up streamed. The user facing interface will remain the same so the current fork should be enough to see what's possible etc.