esp-rs/esp-hal #531: Feature request: make it safe to access `static mut` statics inside `#[entry]` and `#[interrupt]`

dimpolo, 2023-05-11:
The corresponding macros on cortex-m devices allow one to safely access `static mut` statics.
See: https://docs.rs/cortex-m-rt/latest/cortex_m_rt/attr.entry.html#properties

sethp, 2023-05-14:
Neat! That's a cool use of a proc macro to encode an invariant that's otherwise  "invisible" to the compiler. I wonder how it works? I've definitely written bugs arising from accesses to `static mut`s across multiple interrupt handlers (and even one within the same re-entrant interrupt handler), so it seems like it'd have to be more tightly scoped than "any entry point / interrupt handler can access any static."

Ah, it looks like it's not any `static mut` access that get transformed, only the ones local to the function in question. In other words:

```rust
static mut GLOBAL: ... 

#[entry]
fn main() {
    static mut LOCAL: ...
}
```

the cortex-m `#[entry]` macro transforms accesses to `LOCAL` to remove the need for `unsafe` at the point of use, but would not transform `GLOBAL`. Does that match your understanding?

bjoernQ, 2023-05-15:
Nice suggestion! I think I have something like this (at the bottom) of my personal TODO list. Glad I can remove it in favor of a well written issue 👍

dimpolo, 2023-05-15:
@sethp Yes that matches my understanding

I guess someone with more familiarity with the architectures should check if in theory one ISR could run on two cores at once 🤷

DoumanAsh, 2023-06-30:
Due to `noalias` static mut will remain unsound inherently.
If your handler will attempt to take mutable reference to it, your code becomes UB due to `noalias` of this reference.

ideally it should use `UnsafeCell` to wrap mutability and avoid `noalias`