esp-rs/esp-hal #535: Level-triggered GPIO interrupts are firing multiple times

bugadani, 2023-05-15:
[image]

The above is a photo of the following issue:

I have a sensor (ADS1291 in this case), that has a low-active data ready line, and provides 48 bits of data as a response to a read command over SPI.

I also have a read task similar to this:

```rust
loop {
    drdy.wait_for_low().await;
    let sample = sensor.read().await;
    handle(sample);
}
```

It looks like the GPIO interrupt handler is fired multiple times mistakenly. I expected the following order of events:

 - `wait_for_low` registers an interrupt handler and suspends the task
 - data ready goes low, interrupt happens
 - interrupt handler is disabled during interrupt handling
 - sensor read happens, which causes data ready to go high
 - next iteration, `wait_for_low` registers an interrupt handler and starts the waiting game

The issue I'm experiencing here is that the second `wait_for_low` immediately returns, as I hope the photo makes clear.

bugadani, 2023-05-15:
My suspicion is that the order of operations in the [`GPIO` interrupt handler](https://github.com/esp-rs/esp-hal/pull/333/files/3c42b0900c6e0bfc5e7ba21b74e2c570b006b5f0#diff-5cfc259c51ff612ad7e3c9c3e9ea5eb7403058abec590ba9ca6811a0eb59b30cR1727) is incorrect. The driver clears the interrupt status before disabling interrupts, so the hardware seems to re-trigger the interrupt request between clearing and disable. This causes a "misfire" immediately when the interrupt is re-enabled in the next `wait_for_low`.

This is just speculation though that seems to fit my data.

MabezDev, 2023-05-15:
Could you try out https://github.com/esp-rs/esp-hal/pull/537?

PR #537 (2023-05-15): async gpio fixes
- Fix pin number calculation for bank1
- Clear interrupt status after disabling interrupt to avoid hardware pending another interrupt
- Clear interrupt on `InputFuture` creation.

Close #535
closed 2023-05-15 by MabezDev