esp-rs/esp-hal #543: Improve ergonomics of running code on the second core of dual core chips

MabezDev, 2023-05-16:
As mentioned in https://github.com/esp-rs/esp-hal/pull/542, the ergonomics of running code on the second core is not great. I think we should change `FnMut` to `FnOnce`. I tried this but got some hard faults so there is some investigation to do there :D.

MabezDev, 2023-05-16:
Ah, I think we can't just use `FnOnce` as it is because the entry function is stored in core0's stack space. With `FnMut` its not possible for core0 to drop the entry as it (could) mutably borrowed so what we have currently works. When we switch to `FnOnce` core0 can drop the entry function and reclaim the stack space because it thinks the memory is unused (moved to core1) with our current implementation this is not the case.

To switch to `FnOnce` I think we either need to store the start entire function inside a static (might not be possible because `FnOnce` isn't sized?). Or we need to copy the start function into core1's stack space and execute from there.

bugadani, 2023-08-15:
From today's discussion on [matrix](https://matrix.to/#/!LdaNPfUfvefOLewEIM:matrix.org/$Qypt2P17gW1ReCpVqXaRddE4HvrI1tkMgvZaoL4eXjA?via=matrix.org&via=tchncs.de&via=mozilla.org): `embassy-rp` copies the closure to the bottom of the stack: https://github.com/embassy-rs/embassy/blob/main/embassy-rp/src/multicore.rs#L116

closed 2023-08-22 by bjoernQ