esp-rs/esp-hal #561: Dropping a GPIO future does not stop listening for event.

bugadani, 2023-05-25:
Selecting between two async pin events causes the other event to stay registered, which can cause problems down the line (e.g. waking up the UI task that is waiting for the display DMA to finish flushing, which somehow causes my firmware to lock up).

bugadani, 2023-05-26:
I'm not convinced my analysis is correct here. While it's likely that the interrupt handler remains active and that it can wake up the task it's registered to, that wakeup is actually unlikely to cause anything more than just polling the task's future. So what's going on here?

MabezDev, 2023-05-31:
Whilst I can see how it would wake another task up, I'm sure if the DMA transfer hadn't been completed, the poll would return `Poll::Pending` :thinking:.

This is probably something we should implement though, we just have to be careful to `core::mem::forget` the pins used in some drivers (where we just immediately drop them after configuration).

bugadani, 2023-05-31:
I don't think anything should be done with this right now. As long as I can't explain the lockup, nothing is broken except for my logic.

bugadani, 2023-06-01:
I've deleted the incorrect description. I'm leaving the issue open because the title is correct. Feel free to close in case handling the interrupt can be tolerated.

jessebraham, 2024-04-11:
Related: #1382

Dominaezzz, 2024-04-14:
At the very least the future should release the waker. At the moment it's leaked when it's cancelled, which is problematic when the waker was created on the stack.

Sadly the `atomic_waker` in `embassy-sync` doesn't support releasing the waker without waking it. Not a show stopper but a bit awkward to read.
[atomic-waker](https://github.com/smol-rs/atomic-waker) does support it and also doesn't require a critical section (spin locking other core) to work. Might be worth switching over? #1035

MabezDev, 2024-04-14:
> At the very least the future should release the waker. At the moment it's leaked when it's cancelled, which is problematic when the waker was created on the stack.

`AtomicWaker` clones the waker and stores it within itself, so we don't need to release the waker on drop as it will never point to uninitialized memory.

> [atomic-waker](https://github.com/smol-rs/atomic-waker) does support it and also doesn't require a critical section (spin locking other core) to work. Might be worth switching over? https://github.com/esp-rs/esp-hal/issues/1035

This is a very nice idea, I'd be interested in this being explored. It should give us a nice perf boost on multicore chips. It also supports portable-atomic for the chips without atomics.

Dominaezzz, 2024-04-14:
> as it will never point to uninitialized memory.

If the provided waker (that `AtomicWaker` eventually clones) was created on the stack, it will once it goes out of scope. Remember that cloning the waker really just bumps a reference counter but still refers to the same raw waker.

While typing this, I also realised that if the waker implementation is using `Arc` (which the Rust core/alloc implementation does) the waker memory would be leaked. Though the memory will not be uninitialized.

(Just in case you ask) A use case for stack allocation rather than global/heap is for a `select` than minimizes `poll`ing of it's futures.

MabezDev, 2024-04-14:
> If the provided waker (that AtomicWaker eventually clones) was created on the stack, it will once it goes out of scope. Remember that cloning the waker really just bumps a reference counter but still refers to the same raw waker.

The waker comes from the top-level future (task) memory iirc. In the case of embassy, tasks are statically allocated thus the memory will always be valid, but if someone tried to use our async drivers with say [async-task](https://github.com/smol-rs/async-task) which supports dynamic task allocation and deallocation, that could be a problem.

> While typing this, I also realised that if the waker implementation is using Arc (which the Rust core/alloc implementation does) the waker memory would be leaked. Though the memory will not be uninitialized.

The waker impl is up to the executor, what do you mean by "which the Rust core/alloc implementation does"?

> (Just in case you ask) A use case for stack allocation rather than global/heap is for a select than minimizes polling of it's futures.

Is this so that each future has its own stack allocated waker, meaning that when poll is called the future can tell which of the future is ready to make progress without checking through N futures (like `embassy_futures::select` does currently)?

Dominaezzz, 2024-04-14:
> The waker impl is up to the executor, what do you mean by "which the Rust core/alloc implementation does"?

https://doc.rust-lang.org/alloc/task/trait.Wake.html

I use the term "implementation" loosely :sweat_smile:

> Is this so that each future has its own stack allocated waker, meaning that when poll is called the future can tell which of the future is ready to make progress without checking through N futures (like `embassy_futures::select` does currently)?

Yes, precisely that!

MabezDev, 2024-04-14:
Okay, thanks for bringing this to our attention! It seems switching to `atomic-waker` should solve this and we might get a small perf uplift on multicore chips.

MabezDev, 2024-08-07:
We can track this in https://github.com/esp-rs/esp-hal/issues/1382, so closing this.

PR #2625 (2024-12-02): Track async GPIOs in memory
Fixes #561 - even though that issue was closed without resolution.

Drive-by fixes #2654
closed 2024-08-07 by MabezDev