esp-rs/esp-hal #564: Issues with cfg dependency features (probably esp32-hal -> esp-hal-common)

plaes, 2023-05-28:
There seem to be some issues when attempting to build simplest embassy example outside the `esp-hal` as its own project.
How to reproduce:
1. Create a sample project `cargo generate --git https://github.com/esp-rs/esp-template`:
```
🤷   Project Name: esp32-embassy-poc
🔧   Destination: $HOME/esp32-embassy-poc ...
🔧   project-name: esp32-embassy-poc ...
🔧   Generating template ...
✔ 🤷   Use template default values? · true
✔ 🤷   Which MCU to target? · esp32
✔ 🤷   The template is requesting to run the following command. Do you agree?
cargo fmt · yes

Use `cargo run` to flash and run your code

🔧   Moving generated files into: `$HOME/esp32-embassy-poc`...
Initializing a fresh Git repository
✨   Done! New project created $HOME/esp32-embassy-poc
```
2. Copy example from `esp32-hal/examples/embassy_hello_world.rs` to ` $HOME/esp32-embassy-poc/main.rs`
3. Fix the dependencies in the `Cargo.toml`, so project can be built (based on `Cargo.toml` in the esp32-hal directory):
```
...

[dependencies]
#hal = { package = "esp32-hal", version = "0.12.0" }
esp32-hal = { version = "0.12.0", features=["embassy", "embassy-time-timg0"]}
esp-backtrace = { version = "0.7.0", features = ["esp32", "panic-handler", "exception-handler", "print-uart"] }
esp-println       = { version = "0.5.0", features = ["esp32"] }
embassy-executor = { version = "0.2.0", features = ["arch-xtensa", "executor-thread", "integrated-timers", "nightly"] }
static_cell = "1.0.0"
embassy-time = { version = "0.1.1", features = ["tick-hz-1_000_000"] }
```
4. Build the project and run the project on the device:
```
...
Init!
 
 
!! A panic occured in '$HOME/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-hal-common-0.9.0/src/embassy/time_driver_timg.rs', at line 32, column 74
 
PanicInfo {
    payload: Any { .. },
    message: Some(
        called `Option::unwrap()` on a `None` value,
    ),
    location: Location {
        file: "$HOME/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-hal-common-0.9.0/src/embassy/time_driver_timg.rs",
        line: 32,
        col: 74,
    },
    can_unwind: true,
}
 
Backtrace:
 
0x400eec88
0x400eec88 - core::panicking::panic
    at $HOME/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core/src/panicking.rs:114
0x400dd000
0x400dd000 - core::option::Option<T>::unwrap
    at $HOME/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core/src/option.rs:942
0x400df3b4
0x400df3b4 - esp_hal_common::embassy::time_driver::EmbassyTimer::now::{{closure}}
    at $HOME/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-hal-common-0.9.0/src/embassy/time_driver_timg.rs:32
0x400e28d7
0x400e28d7 - critical_section::with
    at $HOME/.cargo/registry/src/github.com-1ecc6299db9ec823/critical-section-1.1.1/src/lib.rs:226
0x400df37d
...
```

plaes, 2023-05-29:
After switching esp32-hal from 0.9.0 to latest git master (https://github.com/esp-rs/esp-hal#a82c9dbc), I get an almost working application, where tasks only run once:
```
Init!
Bing!
Hello world from embassy using esp-hal-async!
```

MabezDev, 2023-05-31:
I fixed this in https://github.com/esp-rs/esp-hal/pull/523, could you try running the example in the hal and see if that works at least?

bowdie, 2023-06-03:
> After switching esp32-hal from 0.9.0 to latest git master (https://github.com/esp-rs/esp-hal#a82c9dbc), I get an almost working application, where tasks only run once:

I was experiencing the original issue that @plaes posted. I switched to the latest git master as suggested, but similarly both task loops run once only.

When `Timer::after(Duration::from_millis(1_000)).await;` is removed it loops as intended (but obviously without the delay.)

plaes, 2023-06-04:
> I was experiencing the original issue that @plaes posted. I switched to the latest git master as suggested, but similarly both task loops run once only.
>
> When [...]

@bowdie Thanks to your input I finally figured it out.. 
In [the example code](https://github.com/esp-rs/esp-hal/blob/34740c7f1c44ee1fc02b4e21b16ed9b3666fe19a/esp32-hal/examples/embassy_hello_world.rs#L67-L68), the timer initialization is behind the feature gate (`embassy-time-timg0`) and in order to build this example, you also need to supply `--features=embassy-time-timg0,embassy`) to cargo.
When you copy the example verbatim to a standalone repository and attempt to build it, `embassy::init(...)` stays behind the feature gate and is not included in compilation:
```
#[entry]
fn main() -> ! {

    ...

    #[cfg(feature = "embassy-time-timg0")]
    embassy::init(&clocks, timer_group0.timer0);

    let executor = EXECUTOR.init(Executor::new());
    executor.run(|spawner| {
        spawner.spawn(run1()).ok();
        spawner.spawn(run2()).ok();
    });
}
```

Which also explains the compile-time warning as well when building this example out-of-tree:
```
warning: unused import: `embassy`
  --> src/main.rs:14:5
   |
14 |     embassy,
   |     ^^^^^^^
   |
   = note: `#[warn(unused_imports)]` on by default
```

@MabezDev - What do you think? Should we close it as "PEBKAC issue", or would it make sense to somehow rework the examples so people won't shoot themselves in the foot in future..?

bowdie, 2023-06-04:
@plaes Thanks very much for spotting that. Confirming it's working as intended taking into account the feature gate.

It was very much Rust noob issue on my end, but I agree that some rakes could be picked up around the examples. Even something as simple as adding some comments above the feature gates indicating that they need to be removed when using in your own code would have been sufficient for a newcomer like me.

MabezDev, 2023-06-04:
> @MabezDev - What do you think? Should we close it as "PEBKAC issue", or would it make sense to somehow rework the examples so people won't shoot themselves in the foot in future..?

Yeah, this is quite the foot gun :D. Glad you figured it out! We can solve this quite easily with a build script check that at least one of the time driver implementations is implemented.

jessebraham, 2023-06-04:
> Yeah, this is quite the foot gun :D. Glad you figured it out! We can solve this quite easily with a build script check that at least one of the time driver implementations is implemented.

We're already checking various features in the build script in `esp-hal-common`, I guess we can add a check that if `embassy` is enabled then exactly one of `embassy-time-timg0` or `embassy-time-systick` are enabled.

jessebraham, 2023-06-15:
Sorry, forgot about this issue. Will try to get it tackled soon.

PR #608 (2023-06-21): If the `embassy` feature is enabled, ensure that a time driver implementation is available
Closes #564
closed 2023-06-21 by MabezDev