esp-rs/esp-hal #579: No hal builds without the `rt` feature enabled.

MabezDev, 2023-06-07:
It should be entirely possible to build the hal without the rt feature, but they currently do not.

MabezDev, 2023-06-07:
It seems the  `rt` feature is always enabled on the pacs, and the runtimes seems to be enabled always. We need to add a feature.

jessebraham, 2023-06-21:
This is actually non-trivial due to the tight coupling between:

- `esp-hal-common` and `esp-riscv-rt` via the `TrapFrame` struct
- `esp-hal-common` and `xtensa-lx-rt` via the `Context` struct

Both of these are required for our interrupt handling.

Other than pulling `TrapFrame`/`Context` out into their own package(s) (which I'm not super excited about doing) I'm not sure how best to approach this.

jessebraham, 2023-07-05:
Ahh yeah you're probably right, didn't even think of that. I also forget why @MabezDev wanted this, so maybe a reminder is in order? 😅

MabezDev, 2023-07-05:
The problem isn't cfging away the interrupt stuff when we don't enable the rt feature, its actually the inverse. When enabling the `rt` feature there doesn't currently seem to be a way to only enable _either_ `esp-riscv-rt` _or_ `xtensa-lx-rt` to get the trap frame. I think whilst we were in Brno we even tried to do it with a build script but that doesn't work because the script is executed after dep resolution.

Maybe we can just enable both? It might hurt build times (not by much at all though, both are very, very small crates) but won't actually do anything unless we include it in our code; which when `rt` is not enabled will be the case. What do you think?

bjoernQ, 2023-07-06:
I think we would try to compile e.g. the RISCV Assembly with the Xtensa target and vice versa, then?
Since we control both `rt` crates we would need to `cfg` - gate that code depending on the `target_arch` to solve that 🤔

MabezDev, 2023-07-10:
Maybe another option I just thought of, we could just push this feature resolution to the actual hal crates, instead of having the logic inside esp-hal-common. I.e `esp32c3-hal` would enable the `esp32c3` feature of esp-hal-common, but also `esp-riscv-rt`, same goes for the other chips/archs.

PR #1057 (2024-01-03): Add `rt-riscv` and `rt-xtensa` features to `esp-hal-common`
Not an ideal solution, but it's not immediately obvious how we would unify these features to me. Happy to make any changes if anybody has a better solution. This should be enough to close #579 at least, though.

Once this gets merged I can rebase #1049 on top of it and wrap that up, too.
closed 2024-01-03 by jessebraham