esp-rs/esp-hal #582: Adding descriptors to a DMA transfer

BryanKadzban, 2023-06-09:
In my application, I'd like to set up continuous DMA over the SPI peripheral, to minimize latency (the setup uses SPI to send and receive wifi frames, so lower latency in the DMA means lower latency in network packet handling).

This means I'd like to be able to modify the transfer while it's happening. The hardware supports a "restart", where you can add more descriptions to the end. It also supports poking at how far along the DMA engine has made it - the DMA controller will turn each descriptor's owner bit from DMA to CPU as it finishes using each descriptor. So it would be possible to poll the owner bits, and add more descriptors once enough of them get used.

But I'm not sure what that API should look like. Would it be safe to add a couple methods to the DmaTransferRxTx trait? The memory safety, in particular, might be off.

Doing this might also require moving the ownership of the descriptor lists - right now they're part of the DMA channel, not the transfer, so extending them at runtime might be hard.

Any thoughts?

bjoernQ, 2023-07-19:
I guess for SPI continuous DMA won't be possible because the peripheral relies on DLEN for the transfer length ( https://github.com/esp-rs/esp-hal/blob/709dae439b8adfa3c2a235e5ac601b159a9f454c/esp-hal-common/src/spi.rs#L2745 )

For peripherals which support continuous DMA there is a way to push more data (e.g. for I2S https://github.com/esp-rs/esp-hal/blob/709dae439b8adfa3c2a235e5ac601b159a9f454c/esp32c3-hal/examples/i2s_sound.rs#L140 )

BryanKadzban, 2023-08-17:
For an SPI master, yeah, it requires the length. But a slave can run for as long as the master wants to keep sending clock pulses.

But that also explains why I hadn't seen the push API - it can't be done on the SPI master, and I wasn't reading the I2S code. Maybe I'll have to add that to the SPI slave once it's

BryanKadzban, 2023-08-17:
Turns out this is possible today, and already done on some peripherals, see above.

bjoernQ, 2023-08-17:
I see ... for an SPI slave something like this might be useful.  Having push/pull functionality would definitely be the better solution.

Is there any reason to not close this issue?

bjoernQ, 2023-08-30:
Closing this issue. Feel free to reopen if you think it should stay open

closed 2023-08-30 by bjoernQ