esp-rs/esp-hal #583: SmartLED adapter sending incorrect data

seanaye, 2023-06-12:
I have modified the `hello_rgb.rs` example for the esp32c6 to the following.

```rs
#![no_std]
#![no_main]

use esp32c6_hal::{
    clock::ClockControl,
    peripherals,
    prelude::*,
    pulse_control::ClockSource,
    timer::TimerGroup,
    Delay,
    PulseControl,
    Rtc,
    IO,
};
use esp_backtrace as _;
use esp_hal_smartled::{smartLedAdapter, SmartLedsAdapter};
use smart_leds::{
    brightness,
    gamma,
    hsv::{hsv2rgb, Hsv},
    SmartLedsWrite,
};

#[entry]
fn main() -> ! {
    let peripherals = peripherals::Peripherals::take();
    let mut system = peripherals.PCR.split();
    let clocks = ClockControl::boot_defaults(system.clock_control).freeze();

    // Disable the watchdog timers. For the ESP32-C6, this includes the Super WDT,
    // and the TIMG WDTs.
    let mut rtc = Rtc::new(peripherals.LP_CLKRST);
    let timer_group0 = TimerGroup::new(
        peripherals.TIMG0,
        &clocks,
        &mut system.peripheral_clock_control,
    );
    let mut wdt0 = timer_group0.wdt;
    let timer_group1 = TimerGroup::new(
        peripherals.TIMG1,
        &clocks,
        &mut system.peripheral_clock_control,
    );
    let mut wdt1 = timer_group1.wdt;

    // Disable watchdog timers
    rtc.swd.disable();
    rtc.rwdt.disable();
    wdt0.disable();
    wdt1.disable();

    // Configure RMT peripheral globally
    let pulse = PulseControl::new(
        peripherals.RMT,
        &mut system.peripheral_clock_control,
        ClockSource::APB,
        0,
        0,
        0,
    )
    .unwrap();

    // We use one of the RMT channels to instantiate a `SmartLedsAdapter` which can
    // be used directly with all `smart_led` implementations
    let io = IO::new(peripherals.GPIO, peripherals.IO_MUX);
    let mut led = <smartLedAdapter!(10)>::new(pulse.channel0, io.pins.gpio8);

    // Initialize the Delay peripheral, and use it to toggle the LED state in a
    // loop.
    let mut delay = Delay::new(&clocks);

    let mut color = Hsv {
        hue: 0,
        sat: 255,
        val: 255,
    };
    let mut data;

    loop {
        // Iterate over the rainbow!
        for hue in 0..=255 {
            color.hue = hue;
            // Convert from the HSV color space (where we can easily transition from one
            // color to the other) to the RGB color space that we can then send to the LED
            data = [
                hsv2rgb(color),
                hsv2rgb(color),
                hsv2rgb(color),
                hsv2rgb(color),
                hsv2rgb(color),
                hsv2rgb(color),
                hsv2rgb(color),
                hsv2rgb(color),
                hsv2rgb(color),
                hsv2rgb(color),
            ];
            // When sending to the LED, we do a gamma correction first (see smart_leds
            // documentation for details) and then limit the brightness to 10 out of 255 so
            // that the output it's not too bright.
            led.write(brightness(gamma(data.iter().cloned()), 10))
                .unwrap();
            delay.delay_ms(20u8);
        }
    }
}

```

I am compiling and flashing with the release flag also as suggested in #355 

`cargo espflash flash --release --format direct-boot --features direct-boot --example hello_rgb --monitor`

The LEDs are not rendering all the same colors despite being all the same values in code.
[image]

SergioGasquez, 2023-06-14:
Hi! Just tried to reproduce the issue and couldn't, I've only used 4 LEDs since I don't have access to more of them, so, do you mind trying with only 4 of them? I've used your code and changed the `smartLedAdapter!` macro to `4` and adjusted the `data` array.

Also, do you happen to know which model of LEDs you are using? I tried it with WCMCU-28

jessebraham, 2023-07-27:
We've updated `esp-hal-smartled` in #694 to use the new `rmt` driver instead of the previous `pulse_control` driver. I unfortunately don't have the time (or hardware) to test this right now, so if somebody is able to try again with the latest `main` branch that'd be much appreciated. Hopefully the issue has resolved itself 😁

jessebraham, 2023-08-22:
I'm going to assume this is fixed by the new driver, since I've received no feedback. Feel free to re-open this issue or to open a new one if the issue has not been resolved.

closed 2023-08-22 by jessebraham