esp-rs/esp-hal #584: ESP32 Embassy Executor on own project panics

StefanB7, 2023-06-12:
Hi.

I'm having issues running the `embassy_hello_world` example in my own project, running the example from the repo works fine with `cargo run --example embassy_hello_world --features=embassy,async,embassy-time-timg0`.

However,  when trying to run the same program in my own project with `Cargo.toml` as below, it results in a panic.

Cargo.toml:
```toml
[package]
name = "hello_world_embassy"
version = "0.1.0"
authors = ["Stefan"]
edition = "2021"
license = "MIT OR Apache-2.0"

[dependencies]
embassy-time       = { version = "0.1.1",  features = ["nightly"] }
embassy-executor  = { version = "0.2.0", features = ["nightly", "integrated-timers", "arch-xtensa", "executor-thread"] }
hal = { package = "esp32-hal", version = "0.12.0", features = ["embassy", "async", "embassy-time-timg0"]}
esp-backtrace = { version = "0.7.0", features = ["esp32", "panic-handler", "exception-handler", "print-uart"] }
esp-println       = { version = "0.5.0", features = ["esp32"] }
static_cell       = "1.0.0"
```

This is the stacktrace:
```
I (201) boot: Loaded app from partition at offset 0x10000
I (201) boot: Disabling RNG early entropy source...
Init!
 

!! A panic occured in 'C:\Users\stefa\.cargo\registry\src\index.crates.io-6f17d22bba15001f\esp-hal-common-0.9.0\src\embassy\time_driver_timg.rs', at line 32, column 74

PanicInfo {
    payload: Any { .. },
    message: Some(
        called `Option::unwrap()` on a `None` value,
    ),
    location: Location {
        file: "C:\\Users\\stefa\\.cargo\\registry\\src\\index.crates.io-6f17d22bba15001f\\esp-hal-common-0.9.0\\src\\embassy\\time_driver_timg.rs",
        line: 32,
        col: 74,
    },
    can_unwind: true,
}

Backtrace:

0x400f0557
0x400f0557 - core::panicking::panic
    at C:\Users\stefa\.rustup\toolchains\esp\lib\rustlib\src\rust\library\core\src\panicking.rs:117
0x400dca04
0x400dca04 - core::option::Option<T>::unwrap
    at C:\Users\stefa\.rustup\toolchains\esp\lib\rustlib\src\rust\library\core\src\option.rs:950
0x400e2c80
0x400e2c80 - esp_hal_common::embassy::time_driver::EmbassyTimer::now::{{closure}}
    at C:\Users\stefa\.cargo\registry\src\index.crates.io-6f17d22bba15001f\esp-hal-common-0.9.0\src\embassy\time_driver_timg.rs:32
0x400e2c2f
0x400e2c2f - critical_section::with
    at C:\Users\stefa\.cargo\registry\src\index.crates.io-6f17d22bba15001f\critical-section-1.1.1\src\lib.rs:226
0x400e2c49
0x400e2c49 - esp_hal_common::embassy::time_driver::EmbassyTimer::now
    at C:\Users\stefa\.cargo\registry\src\index.crates.io-6f17d22bba15001f\esp-hal-common-0.9.0\src\embassy\time_driver_timg.rs:32
0x400e2f13
0x400e2f13 - esp_hal_common::embassy::<impl embassy_time::driver::Driver for esp_hal_common::embassy::time_driver::EmbassyTimer>::now
    at C:\Users\stefa\.cargo\registry\src\index.crates.io-6f17d22bba15001f\esp-hal-common-0.9.0\src\embassy\mod.rs:48
0x400e2f88
0x400e2f88 - _embassy_time_now
    at C:\Users\stefa\.cargo\registry\src\index.crates.io-6f17d22bba15001f\embassy-time-0.1.1\src\driver.rs:157
0x400e8105
0x400e8105 - embassy_time::driver::now
    at C:\Users\stefa\.cargo\registry\src\index.crates.io-6f17d22bba15001f\embassy-time-0.1.1\src\driver.rs:127
0x400e8081
0x400e8081 - embassy_time::instant::Instant::now
    at C:\Users\stefa\.cargo\registry\src\index.crates.io-6f17d22bba15001f\embassy-time-0.1.1\src\instant.rs:21
0x400e73c8
0x400e73c8 - embassy_executor::raw::SyncExecutor::poll
    at C:\Users\stefa\.cargo\registry\src\index.crates.io-6f17d22bba15001f\embassy-executor-0.2.0\src\raw\mod.rs:409
``` 

I am including the `embassy-executor` and `embassy-time ` as they are in the repo's `Cargo.toml` (the `embassy-executor` is added as a dev-dependency in the repo). I also enabled the "embassy", "async" and "embassy-time-timg0" features on the hal crate.

When building my project I also get a weird unused import warning:

```
warning: unused import: `embassy`
  --> src\main.rs:14:5
   |
14 |     embassy,
   |     ^^^^^^^
   |
   = note: `#[warn(unused_imports)]` on by default

warning: `hello_world_embassy` (bin "hello_world_embassy") generated 1 warning (run `cargo fix --bin "hello_world_embassy"` to apply 1 suggestion)
```

my imports for hal looks just like the example in the repo (all of the code does, in fact):
```
use hal::{
    clock::ClockControl,
    embassy,
    peripherals::Peripherals,
    prelude::*,
    timer::TimerGroup,
    Rtc,
};
```

Am I doing something wrong?
Thanks for helping out :-)

MabezDev, 2023-06-12:
This looks like a duplicate of https://github.com/esp-rs/esp-hal/issues/564. It looks like for whatever reason the timer feature isn't being enabled. Also I would use the git main branch as I did some fixes to the timg0 time driver.

StefanB7, 2023-06-12:
Thanks @MabezDev. Yes it is a duplicate.
Removing `#[cfg(feature = "embassy-time-timg0")]` (as per #564) on line 67 solved my problem.
Thanks again

MabezDev, 2023-06-12:
Glad you got it working :). Closing.

closed 2023-06-12 by MabezDev