esp-rs/esp-hal #604: embassy examples panic for risk

alexiionescu, 2023-06-21:
Steps to reproduce:
1. Enter esp32c3-hal subfolder
2. run `cargo run --example embassy_serial --features embassy,async,embassy-time-systick` or `cargo run --example embassy_serial --features embassy,async,embassy-time-systick`. You can run embassy_hello_world and the result is the same.
3. Program panics when loading time driver:
```
!! A panic occured in '/Users/poltys/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embassy-executor-0.2.0/src/raw/mod.rs', at line 355, column 30

PanicInfo {
    payload: Any { .. },
    message: Some(
        unwrap of `driver::allocate_alarm()` failed: NoneError,
    ),
    location: Location {
        file: "/Users/poltys/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embassy-executor-0.2.0/src/raw/mod.rs",
        line: 355,
        col: 30,
    },
    can_unwind: true,
}
```

MabezDev, 2023-06-21:
Could you try this PR and report back? https://github.com/esp-rs/esp-hal/pull/605

jessebraham, 2023-06-22:
I was able to reproduce this panic, using the [fix-riscv-trap-sections](https://github.com/esp-rs/esp-hal/tree/fix-riscv-trap-sections) branch resolves this on my end.