esp-rs/esp-hal #613: FFT HW Acceleration

ramtej, 2023-06-22:
I am currently benchmarking some DSP routines on the ESP32 S3 platform in Rust. Several issues have already arisen, see https://github.com/esp-rs/rust/issues/180.

Upon reading the 'ESP32-S3 Technical Reference Manual', it became apparent that the S3 platform implements some SIMD as well as DSP operations in hardware, such as `EE.FFT.R2BF.S16` or `EE.CMUL.S16`. I would be willing to invest some time and and implement a hardware-accelerated FFT. I am more of a mathematician and do not know the ESP32 architecture well enough, so I need some support.

Would it therefore be possible for someone to guide me and show me where I need to start? I looked at the SHA HW acceleration and can understand most things, but not all.

Thanks, Jiri

ramtej, 2023-06-22:
To me, it seems that the EPS DSP  "C" examples are merely ASM optimized  and do not use any DSP functions!? Is that the case?

See https://github.com/espressif/esp-dsp/blob/71514173b58b960173b40c4ade9d15d372770a74/modules/fft/float/dsps_fft2r_fc32_ae32_.S#L60

igrr, 2023-06-22:
Since these extra instructions are all integer instructions, only fixed-point FFT uses them: https://github.com/espressif/esp-dsp/blob/71514173b58b960173b40c4ade9d15d372770a74/modules/fft/fixed/dsps_fft2r_sc16_aes3.S

ramtej, 2023-06-22:
Yes, that makes sense. This means that the DSP primitives have already been spotted in the wild and not only described in the above paper.  

My approach now would be to fork or extend e.g. https://gitlab.com/teskje/microfft-rs that similar to the SHA HW acceleration use the DSP primitives. Specifically, I would want to replace the radix-2 butterfly computation with the DSP HW functions.

ramtej, 2023-06-23:
Another (maybe stupid) idea could be to use the inline assembler asm!{} macro. Especially because the S3 target supports beside the DSP also some nice SIMD operations. 

The follow complies and even runs on the xtensa platform:

```
    std::hint::black_box(unsafe {
        asm!("nop");
    });
```
Now there is a lot between the above code and the universe that finally runs my code, especially the LLVM. I have dug into the esp Rust fork a bit and it looks like at least something was done in that direction:

```
pub enum InlineAsmArch {
    X86,
    X86_64,
    Arm,
  ..
    Xtensa,
  ..
}
```
https://github.com/esp-rs/rust/blob/ed3726ba7aea45731260ec8f75f05fc60c2b0f22/compiler/rustc_target/src/asm/mod.rs#L201

Is that enough? I had naively just used a random DSP mnemonic - the error message is "mnemonic unknown". 

Does anyone have an idea on the topic?

Thanks, 
Jiri

zRedShift, 2023-07-06:
@ramtej AFAIK, LLVM plain old doesn't support/recognize the hundreds of DSP instructions, so to use inline assembly it would require either adding them all, or using the escaped binary opcode (but that also means specifying registers by number)

I wish support could be added, but it's a non-trivial amount of search and replace in the llvm-project codebase.

ramtej, 2023-07-09:
@zRedShift Is the ".byte 0x00, 0x30, 0x00" sequence an inline assembly instruction specified as a sequence of bytes, serving as a workaround when the desired assembly instruction isn't supported by the LLVM? Indeed, this is a neat trick; I'll give it a try.

Thanks!

ramtej, 2023-07-09:
There is a nice crate that does the '.byte' encoding for the RISC-V V extension instructions - https://github.com/cryptape/rvv-encoder. It would be exciting to have something similar for the Xtensa extension instructions. 

```
unsafe {
    xtensa_asm::asm!(
         ..
        "ee.cmul.s16	q3,q2,q1,3",
        ..
    );
}
```

zRedShift, 2023-07-09:
@ramtej awesome find. This will also help with ESP32P4's custom RISC V DSP extensions, when it comes out.

I haven't investigated it yet (I was actually planning on just writing `.S` files and compiling/linking them), but I wonder, is there support for `rur.*/wsr.*` etc instructions in llvm (chapter 1.6.10 in the ESP32S3 technical reference, Processor Control Instructions). They are neccessary to manipulate special registers that control stuff like FFT width in fixed point mode.

zRedShift, 2023-07-16:
I've been warming up with some Xtensa LLVM backend contributions over the last few days. My end goal is fast (scalar or vector) DSP, for ESP32S3 in Rust, since I'm working on real time audio processing/encoding and I need to maximize the performance to cram in as much processing into the pipeline. This is obviously DCT/FFT/FIR etc. heavy, among other things.

So a few more scalar instruction PRs, and I will move on to adding the 128-bit registers and instructions, so that the inline assembly can be supported directly. It's been a dream of mine for about a year to do it, at the time the ESP32S3 techinical manual still didn't include the extensions, and reverse engineering `bfd/xtensa-modules.c` was a major pain, and by the time they released the instructions in the reference, the priorities shifted. Now is the time to do the work.

Maybe even add auto-vectorization support in the (far) future, but it's a daunting task since the instructions are pipelined, the cost tables need to be populated, and there's a huge amount of user registers that control the runtime behavior of the instructions.

igrr, 2023-07-16:
Just in case, cc @sstefan1 who is planning to merge the initial support for ESP32-S3 DSP instructions into Espressif's LLVM fork soon.

zRedShift, 2023-07-16:
Don't want to step on anyone's toes, if @sstefan1 has already started work on this, I won't pursue, unless I can somehow assist?

sstefan1, 2023-07-17:
Currently we have all ESP32-S3 DSP instructions implemented in LLVM. All instructions are available in clang through clang's builtins, which translate to llvm intrinsics and then to appropriate instructions.

For example:
`__builtin_xtensa_ee_vld_128_ip(1, data, 0);  -->  ee.vld.128.ip  q1, a9, 0`

This work should be merged soon.

I need to investigate how that should be done in rust, though. If anybody already knows, please let me know.

zRedShift, 2023-07-17:
@sstefan1 Well, we don't currently have any xtensa intrinsics support, but it would go to stdarch/core_arch [here](https://github.com/rust-lang/stdarch/tree/master/crates/core_arch). It lives out of tree, so will need to be forked by the `esp-rs` org and the `.gitmodules` at [esp-rs/rust](https://github.com/esp-rs/rust) should point to it. Then they can be added just like in clang. This can be done separately, whenever, and not a blocker for initial support.

For inline assembly support, it's much simpler since most of the base work has already been done by @MabezDev [here](https://github.com/esp-rs/rust/commit/ba3abba8c3097bf023ad7323c129d00196e454f0)
All that needs to be done is add the `qregs` support, and the user regs (`FFT_WIDTH`, `QACC_H_0`, etc.).

If you can get a branch/PR running on [espressif/llvm-project](https://github.com/espressif/llvm-project), I can start work on initial support/testing on Rust for this. I'm already working on those files, adding rust support for the `clamps/minmax` features based on [this](https://github.com/espressif/llvm-project/pull/78) PR.

sstefan1, 2023-07-21:
I've merged the initial support for ESP32S3 DSP instructions in llvm. Just to keep in mind, builtins support is not yet very well tested. I will be doing testing in the following weeks.

sstefan1, 2023-07-21:
One more note, llvm-objdump currently doesn't work correctly with DSP instructions. To check generated assembly, it is best to use llc.

zRedShift, 2023-07-22:
[Here](https://github.com/zRedShift/rust-esp/tree/esp32s3-dsp)'s my branch with experimental support of this in Rust.
I ran into some issues/funky business (with the immediate addressing constant in `ee.vld.128.ip`, which I think is an issue with llvm, since the constant is correct in the generated llvm ir, I'll investigate later), but the core of it works.

zRedShift, 2023-07-22:
@sstefan1 
```llvm
declare void @llvm.xtensa.ee.vld.128.ip(i32, i32, i32) nounwind
define void @test2(i32 %p){
    tail call void @llvm.xtensa.ee.vld.128.ip(i32 5, i32 %p, i32 16)
    ret void
}
```
If I generate assembly with `llc` I get the correct assembly:
```assembly
// llc -O1 -mtriple=xtensa -mcpu=esp32s3 < xtensa-s3-ee-vld-128-ip.ll                                                                                                                                                                                                                    
        .text
        .file   "<stdin>"
        .global test2                           # -- Begin function test2
        .p2align        2
        .type   test2,@function
test2:                                  # @test2
        .cfi_startproc
# %bb.0:
        entry   a1, 32
        .cfi_def_cfa_offset 32
        ee.vld.128.ip    q5, a2, 16
        retw.n
.Lfunc_end0:
        .size   test2, .Lfunc_end0-test2
        .cfi_endproc
                                        # -- End function
        .section        ".note.GNU-stack","",@progbits
```
But if I generate and object file with llc and run gcc objdump (`xtensa-esp32s3-elf-objdump`) I get an issue:
```
// llc -O1 -mtriple=xtensa -mcpu=esp32s3 -filetype=obj < xtensa-s3-ee-vld-128-ip.ll > test.o
// xtensa-esp32s3-elf-objdump -D test.o
test.o:     file format elf32-xtensa-le

Disassembly of section .text:

00000000 <test2>:
   0:   004136          entry   a1, 32
   3:   a39024          ee.vld.128.ip   q5, a2, 0x100
   6:   f01d            retw.n
  ```
  We get 0x100 instead of 0x10, and this happens to all other numbers, only 0 is unaffected.

sstefan1, 2023-07-23:
I will have to look into it. BTW, I'm not sure if xtensa-esp32s3-elf-objdump disassembles DSP instructions correctly either. I had some problems while testing. I will check the llvm as well, but just mentioning I had issues with disassembling too.

zRedShift, 2023-07-23:
Sure, I didn't trust the objdump blindly, I made sure I tested it on one of my ESPS3S3 chips. I ran this memcpy test:
```rust
#[repr(align(16))]
pub struct AlignedArray<const N: usize>([u8; N]);

#[inline(never)]
pub unsafe fn aligned_memcpy_test<const N: usize>(dst: &mut AlignedArray<N>, src: &AlignedArray<N>) {
    let src_addr = src.0.as_ptr();
    let dst_addr = dst.0.as_mut_ptr();
    assert!(src_addr.is_aligned_to(16));
    assert!(dst_addr.is_aligned_to(16));
    assert_eq!(N % 32, 0);
    for _ in 0..N / 32 {
        core::arch::asm!(
            r#"
                ee.vld.128.ip    q0,  {src_addr},  16
                ee.vld.128.ip    q1,  {src_addr},  16
                ee.vst.128.ip    q0,  {dst_addr},  16
                ee.vst.128.ip    q1,  {dst_addr},  16
            "#,
            src_addr = in(reg) src_addr,
            dst_addr = in(reg) dst_addr,
        );
    }
}
```
src: [0, 1, 2, 3, 4, 5, 6, 7, ...., 0, 1, 2, 3, 4, 5,...]
dst: [0, 1, 2, 3, ..., 14, 15, 0, 0, 0 ..., 0, 1, 2, 3, ... , 14, 15, ..., 0, 0, 0, ...]
So the loop is jumping with an offset 320 instead of 32, just like the objdump.

sstefan1, 2023-08-07:
> Sure, I didn't trust the objdump blindly, I made sure I tested it on one of my ESPS3S3 chips. I ran this memcpy test:
>
> ```rust
> #[repr(align(16))]
> pub struct AlignedArray<const N: usize>([u8; N]); [...]

Hi @zRedShift, I was on vacation and wasn't able to look at this earlier. LLVM backend was encoding the `imm16` offset as the actual immediate value, but it should actually encode the multiple of 16. So for 32 it should encode `0x02` for 48 it should encode `0x03` and so on. I will post a fix internally and we should have it on the github repo soon.

zRedShift, 2023-08-07:
@sstefan1 Thank you. I suspected something like that but didn't have time to look into it over the last few weeks. Glad it's been resolved.

I remember also encountering that it was impossible to use `loop/loopnez/loopgtz` in the inline assembly, but since it's not related to the DSP instructions, and hardware loops are already planned to be fixed/included in the future, I didn't investigate it further.