esp-rs/esp-hal #617: proc macro `entry` not expanded: cannot find proc-macro server

johnnybakker, 2023-06-23:
I'm getting this error from rust analyzer in VSCode and Vim.

[image]

jessebraham, 2023-06-24:
Thanks for opening this issue!

This is actually not an issue with the HAL itself, but with how we are building and distributing the `esp` toolchain (with Xtensa support).

I have opened an issue in the `esp-rs/rust` repository requesting that the `rust-analyzer-proc-macro-srv` component is included in our distribution: https://github.com/esp-rs/rust/issues/183

I will leave this issue open in the meantime just so that others who are facing the same problem can see it, but would prefer for any additional conversation to happen in the linked issue above, please 😁

bugadani, 2023-06-26:
Until the issue is resolved, you can see a workaround in [this repo](https://github.com/card-io-ecg/card-io-fw). In essence, I don't set the esp toolchain as default, and I have a set of xtasks to build/flash/etc using the correct toolchain.

jessebraham, 2023-06-26:
Alternatively you can just disable the diagnostic entirely in `.vscode/settings.json`:

```json
{
  "rust-analyzer.diagnostics.disabled": ["unresolved-proc-macro"]
}
```

jessebraham, 2023-07-05:
This issue has been resolved via [esp-rs/rust-build#215](https://github.com/esp-rs/rust-build/pull/215), so the next release of the Xtensa toolchain will contain this fix. Expect this release in the next couple weeks.

I will leave this issue open until the toolchain has been released.

closed 2023-07-18 by bjoernQ