esp-rs/esp-hal #643: Add support for the ESP32-P4

jessebraham, 2023-07-05:
Still no ETA on this chip, however it's next in line.

### Status

- [ ] Create an SVD
- [ ] Create a PAC
- [ ] Add support to ancillary crates (`esp-backtrace`, `esp-println`)
- [ ] Create a HAL package
- [ ] Final review and publication

jessebraham, 2023-12-15:
SVD has been created: https://github.com/espressif/svd/blob/main/svd/esp32p4.svd
PAC has been created: https://github.com/esp-rs/esp-pacs/tree/main/esp32p4

Should have hardware soon, I hope. Once I have acquired a devkit I will try to get a minimal application together so we can finish updating `esp-println`/`esp-backtrace`, and then will PR the minimal `esp32p4-hal` package so that others can collaborate on this.

jessebraham, 2024-01-16:
My devkits have finally arrived, so I have begun work on this.

jessebraham, 2024-01-19:
Initial support added in #1101. Doesn't do anything yet, but serves as a starting point for other work.

jessebraham, 2024-03-20:
This work is being de-prioritized for the time being.

jessebraham, 2024-04-17:
There have been yet more delays with this chip, so we are going to remove the partial support that we currently have (https://github.com/esp-rs/esp-hal/pull/1461) and I am going to close this issue.

Whenever the ESP32-P4 has a release date set in stone we can revisit support and open a new issue.

closed 2024-04-17 by jessebraham

jhaand, 2025-08-16:
Where is the current status on the P4 HAL maintained?