esp-rs/esp-hal #645: Use GPIO19/20 as push pull output

elkel53930, 2023-07-06:
Hi.
I am using ESP32-S3.
I connect LEDs to GPIO19, 20 and 21 and be trying to make them blink by referring to the example.
```Rust
    let io = IO::new(peripherals.GPIO, peripherals.IO_MUX);

    let mut led_g = io.pins.gpio19.into_push_pull_output();
    let mut led_b = io.pins.gpio20.into_push_pull_output();
    let mut led_r = io.pins.gpio21.into_push_pull_output();

    let mut delay = Delay::new(&clocks);

    led_r.set_low().unwrap();
    led_g.set_low().unwrap();
    led_b.set_low().unwrap();

    loop {
        led_r.toggle().unwrap();
        led_g.toggle().unwrap();
        led_b.toggle().unwrap();
        delay.delay_ms(500u32);
    }
```
GPIO21 is blinking without any problem. But GPIO19 and 20 are not blinking.
GPIO19 and 20 are USB ports by default. Do I need to do further configuration?

bjoernQ, 2023-07-06:
For ESP32-C3 we already have some special handling for the USB pins - probably we need something similar for ESP32-S3 then

elkel53930, 2023-07-07:
Thank you for your reply.
I would like to contribute to this ticket. But actually, I am a Rust newbie. This is my first Rust project.
If I try to contribute with code, I am sure I will cause trouble for everyone...

jessebraham, 2023-07-07:
You should just be able to expand on the changes made in this PR: https://github.com/esp-rs/esp-hal/pull/461

I thought I had tested the S3 and didn't run into issues, but it's possible I made a mistake so we can just apply this fix regardless I suppose.

elkel53930, 2023-07-07:
Thanks for the advice!
I will give it a try.

elkel53930, 2023-07-10:
I followed your advice and modified gpio.rs and the blinking works. Thanks!

elkel53930, 2023-07-12:
May I make a PR? Please let me know if there are any rules that I should follow.

jessebraham, 2023-07-12:
No need to ask, contributions are always appreciated! We don't have a contributing guide yet, but there is a pull request template on this repository which will list some guidelines for you.

elkel53930, 2023-07-12:
Understood. And sorry to borther you.
I checked the pull request template.
However, I can not satisfy "All examples work." because I don't have ESP32 except S3 and be using a my original board.
Can I skip some examples? Do you have guidelines?

bjoernQ, 2023-07-13:
Thanks for letting us know you won't be able to check the examples on all targets.

The PR template is more meant as a reminder and the responsibility to do a final check is on the maintainer who will do the review and finally merge it. Would be nice to mention the checked targets in the PR however

elkel53930, 2023-07-19:
Thank you. I made a PR.

bjoernQ, 2023-07-20:
Closed via #663

closed 2023-07-20 by bjoernQ