esp-rs/esp-hal #648: Share an SPI interface in different parts of the code

Skilvingr, 2023-07-10:
Hi, I'm implementing a GC9A01 display using SPI and DMA. Everything works well, but now I need to share the SPI instance in different parts of the code.
To be brief, these are the steps:

1. Create the SPI instance `SPI` with `Spi::new()`;
2. Use it to build a display interface `DI` (crate `display-interface`) (`DI` takes ownership of `SPI`);
3. Pass this last interface to the display builder `Display` (crate `mipidsi`) (`Display` takes ownership of `DI`).

Given these steps, I eventually loose the ownership of the first interface `SPI`, so I can't store it in the object in which I would later make use of it.
I've managed to workaround this issue by creating two different interfaces (the latter of which uses a stolen SPI peripheral, as well as a new DMA channel and separate descriptors); here is part of the code:
```
let (spi, stolen_spi) = unsafe {(
  Spi::new(
      peripherals.SPI2,
      core::ptr::read(&sclk as *const _),
      core::ptr::read(&mosi as *const _),
      core::ptr::read(&dc as *const _),
      core::ptr::read(&cs as *const _),
      80u32.MHz(),
      SpiMode::Mode0,
      &mut system.peripheral_clock_control,
      &clocks
  ).with_dma(dma_channel.configure(
      false,
      &mut descriptors,
      &mut rx_descriptors,
      DmaPriority::Priority0,
  )),
  Spi::new(
      Peripherals::steal().SPI2,
      sclk,
      mosi,
      core::ptr::read(&dc as *const _),
      core::ptr::read(&cs as *const _),
      80u32.MHz(),
      SpiMode::Mode0,
      &mut system.peripheral_clock_control,
      &clocks,
  ).with_dma(dma_channel2.configure(
      false,
      &mut descriptors2,
      &mut rx_descriptors2,
      DmaPriority::Priority0,
  ))
)}
```
Apart from the fact that this causes a colour inversion (and I still cannot figure out why), this would be an ugly workaround anyway.

So my question is: is there a way to share an SPI interface?
And if not so, is it possible to implement such a thing? Or I'm missing something about the protocol?

bjoernQ, 2023-07-11:
Not sure if I really understand your requirements but people commonly use crates like https://crates.io/crates/shared-bus/0.2.5/dependencies or https://crates.io/crates/embedded-hal-bus/0.1.0-alpha.3

Skilvingr, 2023-07-11:
Good to know these crates, thanks!
However my requirement was to obtain two copies of the same SPI interface: one for the display interface, and the other one to start the DMA transaction, in a function apart from the one in which I initialise the interface itself.
`core::ptr:read` seems to do the job.

bjoernQ, 2023-07-11:
Good to hear you found a solution.

Not sure if it might help but I experimented a bit with SPI DMA and DisplayInterface here: https://github.com/bjoernQ/rust-esp32s3-ili9341/blob/main/src/spi_dma_displayinterface.rs

The code is a bit outdated and might be incomplete - but maybe it's an inspiration (i.e. implementing DisplayInterface yourself so you have more control over the SPI peripheral)

Skilvingr, 2023-07-11:
I'll take a look. Thank you very much indeed!

closed 2023-07-11 by Skilvingr