esp-rs/esp-hal #657: ESP32-C3 embassy wait_for not working correctly

anriha, 2023-07-14:
I've been rewriting my old project with embassy, and I've tried using wait_for_rising_edge/wait_for_falling_edge/wait_for_low/wait_for_high, but I can't get them to work. If I check pin state just using is_low/is_high it works as it should, but with wait_for it does completely unpredictable stuff. It is not even consistent, if I restart my device multiple times (without changing anything else) I always get a different result. In the end, I just implemented them myself, but it seems weird to me that those don't work.

The code I use was based on embassy_wait example.

MabezDev, 2023-07-14:
Could you provide the code that produces the issue you are running into please?

anriha, 2023-07-14:
Relevant bits:
```rust
async fn get_magnet_size(magnet_pin: &mut Gpio3<Input<PullUp>>) -> Duration {
    CURRENT_MOTOR_ACTION.store(MotorAction::Backward.into(), Ordering::Relaxed);
    println!("Waiting for start of magnet");
    println!("magnet state {:?}", magnet_pin.is_low());
    magnet_pin.wait_falling_edge().await.unwrap();
    println!("magnet state {:?}", magnet_pin.is_low());
```

```rust
#[entry]
fn main() -> ! {
    println!("Init!");
    let peripherals = Peripherals::take();
    let mut system = peripherals.SYSTEM.split();
    let clocks = ClockControl::boot_defaults(system.clock_control).freeze();

    let mut rtc = Rtc::new(peripherals.RTC_CNTL);
    let timer_group0 = TimerGroup::new(
        peripherals.TIMG0,
        &clocks,
        &mut system.peripheral_clock_control,
    );
    let mut wdt0 = timer_group0.wdt;
    let timer_group1 = TimerGroup::new(
        peripherals.TIMG1,
        &clocks,
        &mut system.peripheral_clock_control,
    );
    let mut wdt1 = timer_group1.wdt;

    // Disable watchdog timers
    rtc.swd.disable();
    rtc.rwdt.disable();
    wdt0.disable();
    wdt1.disable();

    embassy::init(&clocks, timer_group0.timer0);

    let delay = Delay::new(&clocks);
    let io = IO::new(peripherals.GPIO, peripherals.IO_MUX);
    let dout = io.pins.gpio0.into_floating_input();
    let sck = io.pins.gpio1.into_push_pull_output();
    let hx711 = Hx711::new(delay, dout, sck).unwrap();

    let magnet_pin = io.pins.gpio3.into_pull_up_input(); // low if present
    let forward_pin = io.pins.gpio5.into_push_pull_output();
    let backward_pin = io.pins.gpio4.into_push_pull_output();
    let power_pin = io.pins.gpio6.into_push_pull_output();

    hal::interrupt::enable(
        hal::peripherals::Interrupt::GPIO,
        hal::interrupt::Priority::Priority1,
    )
    .unwrap();

    let executor = EXECUTOR.init(Executor::new());
    executor.run(|spawner| {
        // spawner.spawn(weight(hx711)).ok();
        spawner.spawn(emergency_stop()).ok();
        spawner
            .spawn(motor_controller(forward_pin, backward_pin, power_pin))
            .ok();
        spawner.spawn(cleaning_manager(magnet_pin)).ok();
    });
}

```

Doesn't really mater what wait_for I use it just does random things. The println will output a different combination of true/false every time.

bugadani, 2023-07-17:
Just a guess, but Embassy's scheduling is fair, meaning it will poll every task once before re-polling a task. If your magnet input transitions quickly, you may actually miss transitions. You might want to try running `cleaning_magnet` in an interrupt executor for this reason.

anriha, 2023-07-17:
That is something I also considered, but it should be stable for about a half second each time it gets turned on. So I would expect that not to be an issue. Maybe the edge of the state is not stable. Just to be sure, I will try it again with interrupt executor.

anriha, 2023-07-22:
This seems to be fixed with https://github.com/esp-rs/esp-hal/pull/670. Also, the instability with each reset seems to be due to my stupidity. I didn't realize that I need to add delay before reading pins with internal pull up.

bugadani, 2023-07-22:
> that PR still doesn't fix the issue

I'm not surprised, that PR should have minimal visible effects if any at all. Do you have some instruments you can use to debug this issue? It would be nice to see a logic analyzer/oscilloscope graph of this issue: when you wait for an edge transition, write the input state to a GPIO output, and capture the waveforms of both the input and output. This should verify what's happening: how quickly the firmware reacts, and if it reads a correct input value.

> I didn't realize that I need to add delay before reading pins with internal pull up.

RC time constants aren't our friends here :)

anriha, 2023-07-22:
Unfortunately, I don't have an oscilloscope or anything similar. I guess I could do something like connect another ESP and just measure analog input, but not sure if that would be that helpful. Btw, I wanted to try the interrupt executor as you suggested, but it seems that it is not available for riscv, so I was not able to.

> RC time constants aren't our friends here :)

I didn't even think about such issues. I am really not used to things being this fast :)

bugadani, 2023-07-22:
> I didn't even think about such issues

Actually, if the hardware detects a transition, the next read should be stable unless your input is bouncing really fast. Can you share schematics or some details about your hardware?

> I wanted to try the interrupt executor as you suggested, but it seems that it is not available for riscv, so I was not able to.

I made my own for Xtensa, maybe you can adapt it for your firmware: https://github.com/card-io-ecg/card-io-fw/blob/main/src/interrupt.rs

anriha, 2023-07-22:
The relevant part is just a hall sensor connected directly to ESP. Unfortunately, I am unable to know which exact sensor it uses, as I am repairing a cat toilet by replacing the MCU with my own.

>I made my own for Xtensa, maybe you can adapt it for your firmware: https://github.com/card-io-ecg/card-io-fw/blob/main/src/interrupt.rs But right now I'm doubtful if it would help.

Thanks, but I am probably just going to stick with my current solution, by just naively reimplementing wait_for. 

For reference, my reimplementation that works:
```rust
async fn wait_for_rising_edge(magnet_pin: &Gpio3<Input<PullUp>>) {
    let mut last_state = magnet_pin.is_low().unwrap();
    loop {
        let current_state = magnet_pin.is_low().unwrap();
        if last_state && !current_state {
            break;
        }
        last_state = current_state;
        Timer::after(Duration::from_millis(50)).await;
    }
}
```

But I am aware that this is not the same.

bugadani, 2023-07-22:
Your solution is fine if it works for you :) Interrupt executors are relevant when you have some special requirements. For me, I'm measuring a sensor at 1kHz and my stuff just doesn't work without it. But for a magnet that changes maybe once a second, you should be fine.

Still, it would be optimal if we could drill down to the root of your issue.

Can you tell if the hall effect sensor has a decoupling capacitor close to the sensor? They tend to bounce if the sensor is placed on a cable without a capacitor.

anriha, 2023-07-22:
As far as I can tell, no, but it is embedded into the construction of the toilet, so I cannot say for sure. Also, the strange part is that the issue isn't just at the edges of magnet, but even in random places with no magnet nearby. The magnet itself is just to tell where is home position for the rotating body, the entire rotation takes about 2 minutes, and sometimes it just reads a falling/rising edge in the middle of rotation.

MabezDev, 2024-02-19:
Async GPIO, interrupt handling & embassy/timer support have seen several fixes since this issue was originally opened. Can you test again?

anriha, 2024-02-19:
Unfortunately, no change even with updated dependencies. 
```yaml
embassy-executor = { version = "0.5.0", features = ["integrated-timers", "nightly"] }
embassy-time = "0.3.0"
embedded-hal-async = "1.0.0"
esp-backtrace = { version = "0.11.0", features = ["esp32c3", "panic-handler", "exception-handler", "println"] }
esp-println = { version = "0.9.0", features = ["esp32c3", "log"] }
esp32c3 = "0.20.0"
esp32c3-hal = { version = "0.15.0", features = ["async", "embassy", "embassy-executor-thread", "embassy-time-timg0"] }
hx711 = "0.6.0"
log = "0.4.20"
num_enum = { version = "0.7.2", default-features = false }
```

MabezDev, 2024-02-19:
Thanks for testing again. My issue is that I cannot reproduce the issue, therefore I can't look into it. To rule out any hardware discrepancies, could you try and reproduce your issue just using a devkit and modifying one of the examples in esp-hal?

MabezDev, 2024-03-21:
Without being able to reproduce this, I am assuming a hardware issue in your project. Closing for now, please feel free to comment with more info and we can re-open this.

closed 2024-03-21 by MabezDev