esp-rs/esp-hal #699: Disable watchdogs before user code

MabezDev, 2023-07-27:
Due to the ROM bootloader in espressif chips, the watchdog is left enabled when jumping to the main application code. This means we have to disable it in _every single_ example. If we instead disable it before executing user code, we make the watchdog opt-in.

bjoernQ, 2023-08-07:
Another idea I had was to have a helper function to disable the WDTs explicitly but with much less code.

Something like
```rust
    disable_watchdogs(
        &clocks,
        &mut system.peripheral_clock_control,
        &mut peripherals.RTC_CNTL,
        &mut peripherals.TIMG0,
        &mut peripherals.TIMG1,
    );
```

It's certainly still more to do than in the implementation of the original idea here ( https://github.com/esp-rs/esp-hal/pull/704 ) but it's more explicit. Probably it's nicer to have this implicit - just wanted to leave this here

jessebraham, 2023-09-01:
Closed by #763

closed 2023-09-01 by jessebraham