esp-rs/esp-hal #700: RFC: Move esp-wifi into the esp-hal repository

MabezDev, 2023-07-27:
Instead of esp-wifi being all alone, why don't we let it join the mono-repo gang.

### Pros

- We can easily add esp-wifi CI check for modifications to esp-hal so we don't break esp-wifi inadvertently
- We can refactor esp-wifi on the fly, instead of a block refactor when we release new hals, this should hopefully reduce/breakdown the maintenance requirements in esp-wifi

### Cons

- Possibly less discoverability, but we can add redirect before archiving esp-wifi to point to esp-hal
- CI issues, we already have a lot of jobs, this will most likely add more
- Separation of concerns (???) - This is a hand-wavey con, because in reality they are very tightly coupled.

jessebraham, 2023-07-29:
Not a huge fan of this idea to be honest, but we can see what others have to say I guess.

AnthonyGrondin, 2023-08-01:
This will add a huge codebase into `esp-hal` directly, but looking at how `esp-wifi` is codependant on `esp-hal`, this would prevent a lot of breaking changes, when doing changes that require both packages to be update, and also provides better compatibility.

I think `esp-wifi` use proprietary blobs too, while `esp-hal` uses none so far, so that might be something of concern too.

General functionnalities of the Esp services, like:
- Wifi ([esp-wifi](https://github.com/esp-rs/esp-wifi/))
- TLS ([esp-mbedtls](https://github.com/esp-rs/esp-mbedtls/))
- Storage ([esp-storage](https://github.com/esp-rs/esp-storage))
- OTA ([Experiment phase](https://github.com/bjoernQ/esp32c3-ota-experiment))
- DHCP ([Experiment phase](https://github.com/bjoernQ/esp32c3-dhcp-experiment))
- Matter-rs (Experiment phase)

Have been developped in their own crates, so we could entertain the idea of doing what is done with IDF, and have a crate for the hal (`esp-idf-hal`) and one crate for the services (`esp-idf-svc`).

bugadani, 2023-08-01:
> This will add a huge codebase into esp-hal directly

I think the proposal is just to copy the crate into the repository, and not to merge esp-wifi into esp-hal.

I don't have strong opinions here. Specifying esp-hal as a path dependency could make hacking esp-wifi somewhat easier, and it's simpler to make simultaneous changes to both crates, but I don't think these are very important.

bjoernQ, 2023-08-07:
I have mixed feelings about this. It definitely makes some things easier (because of the tight coupling)

The thing I'm a bit concerned of is that GitHub doesn't give us a good way (well ... no way besides using labels) of separating issues and PRs - not sure how much of an issue this is in reality t.b.h. 

On the other hand, making esp-hal break esp-wifi less often would be really nice!

Probably it would also make it a bit less of a pain to add new chip support to esp-wifi since currently we have the radio-clocks stuff here which is used by esp-wifi. And I think there was a proposal to also move the ROM function definitions from esp-wifi to esp-hal

jessebraham, 2023-12-15:
For the time being I'm going to make an executive decision and say "no" to this. I don't think it makes sense right now, we can revisit this in the future if there is a desire or need for it.