esp-rs/esp-hal #705: Can't use embassy-time when the async feature is enabled.

f-edens, 2023-08-05:
First of all: sorry if this isn't an actual issue, but caused by me misunderstanding how cargo's package management works.

I tried to make use of an serial device in a similar way to the [embassy_serial example](https://github.com/esp-rs/esp-hal/blob/main/esp32c3-hal/examples/embassy_serial.rs). However as soon as I enable the async feature and add the embassy-time dependency, the project becomes impossible to build.

Steps to reproduce:

* Create a new project
`> cargo generate esp-rs/esp-template`
  * project name? `foo`
  * advanced template options? `false`
  * mcu? `esp32c3`
* Enter project directory
* add embassy-time dependency
`> cargo add embassy-time`
* Manually add the embassy and the async features to the hal dependency (by editing the Cargo.toml)

Resulting dependencies list:
```
[dependencies]
hal = { package = "esp32c3-hal", version = "0.10.0", features = ["embassy", "async"] }
esp-backtrace = { version = "0.7.0", features = ["esp32c3", "panic-handler", "exception-handler", "print-uart"] }
esp-println       = { version = "0.5.0", features = ["esp32c3"] }
embassy-time = "0.1.2"
```

Now trying to build in any way results in the following error:

`> cargo check`
```
error: failed to select a version for `embedded-hal`.
    ... required by package `esp-hal-common v0.10.0`
    ... which satisfies dependency `esp-hal-common = "^0.10.0"` (locked to 0.10.0) of package `esp32c3-hal v0.10.0`
    ... which satisfies dependency `hal = "^0.10.0"` (locked to 0.10.0) of package `foo v0.1.0 (<redacted>/foo)`
versions that meet the requirements `=1.0.0-alpha.10` are: 1.0.0-alpha.10

all possible versions conflict with previously selected packages.

  previously selected package `embedded-hal v1.0.0-alpha.11`
    ... which satisfies dependency `embedded-hal = "=1.0.0-alpha.11"` (locked to 1.0.0-alpha.11) of package `embedded-hal-async v0.2.0-alpha.2`
    ... which satisfies dependency `embedded-hal-async = "^0.2.0-alpha.1"` (locked to 0.2.0-alpha.2) of package `esp32c3-hal v0.10.0`
    ... which satisfies dependency `hal = "^0.10.0"` (locked to 0.10.0) of package `foo v0.1.0 (<redacted>/foo)`

failed to select a version for `embedded-hal` which could resolve this conflict
```

manio, 2023-08-07:
hi,
In such cases i generaly try to run `cargo tree|grep crate_name`.

I have currently a well tested dependencies which are working but for ESP32. Just try those versions/adjust to your project:
```
[dependencies]
embassy-executor  = { version = "0.2.0", features = ["nightly", "integrated-timers", "arch-xtensa", "executor-thread"] }
static_cell = "1.1.0"
esp-wifi = { git = "https://github.com/esp-rs/esp-wifi", features = ["esp32", "embassy-net", "embedded-svc", "wifi"] }
embassy-time = { version = "0.1.1",  features = ["nightly", "tick-hz-1_000_000"] }
embassy-net = { version = "0.1.0", features = ["nightly", "tcp", "udp", "dhcpv4", "medium-ethernet"] }
hal = { package = "esp32-hal", version = "0.13.0", features = ["embassy", "embassy-time-timg0", "async"] }
esp-backtrace = { version = "0.7.0", features = ["esp32", "panic-handler", "exception-handler", "print-uart"] }
esp-println = { version = "0.5.0", features = ["esp32", "log"] }
embedded-svc = { version = "0.25.3", default-features = false }
embedded-hal-async = { version = "=0.2.0-alpha.1"}
log = "0.4.16"
embedded-io = "0.4.0"
format_no_std = "1.0.0"
```

f-edens, 2023-08-08:
Thank you for your response! It did lead me on the right track towards fixing the issue. `cargo tree` would fail with the same error as `cargo check` does, but because of your dependency list I tried downgrading the `embassy-time` dependency to version `0.1.1`, which fixed the issue

closed 2023-08-08 by f-edens