esp-rs/esp-hal #724: Support interrupt levels 4 and 5

bugadani, 2023-08-15:
@MabezDev you've commented in `CpuInterrupt::level()` the following:

> // we direct these to None because we do not support interrupts at this level
> // through Rust

The [original implementation PR](https://github.com/esp-rs/esp-hal/pull/103) also says that 3 levels are available.

But why? According to their TRMs, the interrupt matrix in ESP32, S2 and S3 has 5 interrupt levels, and at least for level-triggered, all 5 are available. It would be nice if these levels were available for use.

MabezDev, 2023-08-15:
The Xtensa TRM recommended using interrupt levels 4 & 5 as assembly-only routines, that's why it was initially unavailable in the vectored interrupt PR. I think its possible to use them now though xtensa-lx interrupt API, but I've never tried that.

I'm still undecided if we should just ignore the Xtensa TRM and just open up all the interrupt levels to high level Rust code.

bugadani, 2023-08-15:
Is this an either-or situation? The 32 separate interrupt handlers map to the 7 level entry points and that's it?

bjoernQ, 2023-08-30:
I guess we shouldn't

From Xtensa Instruction Set Architecture (ISA) Reference Manual
> The value of CINTLEVEL is set to at least EXCMLEVEL whenever PS.EXCM=1. Thus, all
interrupts at level EXCMLEVEL and below are masked during the time PS.EXCM=1. This
is done to allow high-level language coding with the Windowed Register Option of interrupt handlers for interrupts whose level is not greater than EXCMLEVEL.

For our targets EXCMLEVEL is 3.

My understanding is that we shouldn't have any code at those interrupt levels using the windowed register ABI (it could interfere with the register spilling etc.)

bugadani, 2023-08-30:
I'll be honest, I find this ISA very ... arcane. But it is, what it is 😅

closed 2023-08-30 by bugadani