esp-rs/esp-hal #732: Wrong dram_seg len in linker script for esp32s2-hal

M4tsuri, 2023-08-16:
The `len` field of `dram_seg` is probably wrong:

https://github.com/esp-rs/esp-hal/blob/37e24af753be01748989f3dd8072194864ce63ab/esp32s2-hal/ld/memory.x#L25

The original ending of `dram_seg` is `0x3ffe0000`, which corresponds to `0x40050000` when accessing from the instruction bus. However, the [bootloader already occupies `0x4004F000 ~ 0x4004F000 + 0x7000`](https://github.com/espressif/esp-idf/blob/master/components/bootloader/subproject/main/ld/esp32s2/bootloader.ld#L14). This can lead to the following error when the size of `dram_seg` falls between 188k to 192k:

[image]

(This error is raised [here](https://github.com/espressif/esp-idf/blob/3befd5fff72aa6980514454a50233037718b611f/components/bootloader_support/src/esp_image_format.c#L414))

After some experiments, I believe the correct len should be `188k - RESERVE_CACHES - VECTORS_SIZE`.

bjoernQ, 2023-08-16:
```
components\esp_system\ld\esp32s2\memory.ld.in
35:#define DATA_RAM_END      0x3FFDF000  /* 2nd stage bootloader iram_loader_seg starts at end of block 13 (reclaimed after app boots) */

components\esp_system\ld\esp32s2\memory.ld.in
33:#define RAM_DRAM_START    0x3FFB0000
```

Which is 188k (minus CACHE_SIZE, minus VECTORS_SIZE) 👍

M4tsuri, 2023-08-16:
Thank you for confirming, I'm going to open a PR for this.

M4tsuri, 2023-08-16:
Closed in favor of #734

closed 2023-08-16 by M4tsuri