esp-rs/esp-hal #741: HW SHA Calculation Error in esp32c3-hal

Ethan-Archer, 2023-08-18:
I'm currently working on a project based on the ESP32-C3 platform and utilizing the esp-wifi library. As per the library's requirements, I've added link-arg=-Trom_functions.x to the /.cargo/config.toml file. However, I've encountered an unexpected issue: the Hardware-accelerated SHA (HW SHA) calculation results from the esp32c3-hal library are incorrect after making this modification.

What's puzzling is that when applying the same configuration, specifically link-arg=-Trom_functions.x, to the ESP32-S3 platform, the Hardware-accelerated SHA calculation works as expected without any errors.

To diagnose this, I've also tried using the code provided in the https://github.com/esp-rs/esp-hal/blob/main/esp32c3-hal/examples/sha.rs example, and the issue still persists.

Could you please help me understand why the Hardware-accelerated SHA calculation is encountering errors in ESP32-C3, while ESP32-S3 does not exhibit this issue under the same configuration? Is there something I might be overlooking or not handling correctly in my setup?

bjoernQ, 2023-08-18:
Is adding `link-arg=-Trom_functions.x` to the `/.cargo/config.toml file` (and adding the dependency) already enough to trigger this problem or is it also needed to initialize esp-wifi or even maybe be connected to WiFi?

If just linking `rom_functions.x` is enough then that's very surprising since apparently the SHA driver code doesn't use any ROM functions

Ethan-Archer, 2023-08-18:
Only add to config.toml file causes the issue, esp-wifi initialization is not necessary.

bjoernQ, 2023-08-18:
Ok this is wild!

I was able to track it down to:
```
memset = 0x40000354;
``` 

This will replace Rust's `memset` compiler-builtin function with the ROM function. They _should_ do the exact same thing but apparently, they don't. 🤯

Most probably we can remove those things from esp-wifi's `rom_functions.x` but we need to carefully test things

Still, I wonder how exactly the implementation differ