esp-rs/esp-hal #757: [ESP32] RMT transmission with data.len() >= 72 produces incorrect output

wrazik, 2023-08-26:
Hi! 
I'm trying to create driver for SK6812 LED. However, when I'm trying to control more than 2 LED, some weird color pattern happens. I debug it using salae-logic analyzer, and it turns out that RMT sends incorrect data. 
```rust
// . . .
let mut channel = rmt
    .channel0
    .configure(
        io.pins.gpio25,
        TxChannelConfig {
            clk_divider: 1,
            idle_output: true,
            idle_output_level: false,
            ..TxChannelConfig::default()
        },
    )
    .unwrap();
// . . . 
let mut data = [PulseCode {
    level1: true,
    length1: 100,
    level2: false,
    length2: 50,
}; 72];    // <-------- 71 works normally!!!

data[data.len() - 1] = PulseCode::default();

let transaction = channel.transmit(&data);
```
Following code, instead of sending 71 pulses, sends 135. Sending 70 pulses (data.len() == 71), it works as expected. 

[image]

bjoernQ, 2023-08-28:
Works for me on current main (`91b9b51426302e86c967a130611c7338ae3ad049`)

Even for debug builds (I suspected you are using debug builds which might be too slow to refill the transmit buffer ... but even that works for me)

wrazik, 2023-08-29:
I confirm - building in release mode works as expected!

bjoernQ, 2023-08-30:
Good to hear it's working in release mode. Unfortunately, in develop mode Rust is too slow and it's not really expected to work. (Although it did for me in my tests). Maybe you have a few frequent interrupts in your application?

Since it's working as intended I guess we can close this?

closed 2023-08-30 by jessebraham