esp-rs/esp-hal #766: Period calculation for `TIMG` watchdog timers on C2/H2 is incorrect

jessebraham, 2023-08-30:
While the `watchdog` examples *do* work, they seem to take ~2 times as long as expected to reset (the other chips print `hello world` on average twice before resetting, where as these two devices print it on average 4 times).

bjoernQ, 2023-08-31:
Spend quite some time with this. Doesn't make any sense to me

As a workaround we could just half the `timeout_raw` for these targets. But that's of cause not a satisfying solution

MabezDev, 2024-02-19:
I'm wondering if this was/is a xtal frequency issue? :thinking:

MabezDev, 2024-02-29:
I think this was! Running the `rtc_watchdog` example now it resets after 5 second on H2 (I don't have a 26mhz C2 to test, but I assume its the same). Closing.

closed 2024-02-29 by MabezDev