esp-rs/esp-hal #775: Find an alternative for the `dangoslen/changelog-enforcer@v3` action

jessebraham, 2023-09-01:
This has been driving me insane for months now, it needs to be replaced.

bjoernQ, 2023-09-21:
Is it just this issue: https://github.com/dangoslen/changelog-enforcer/issues/176 ?

Or are the other things we want to be different?

jessebraham, 2023-09-21:
It's mostly just the fact that if you don't add the `skip changelog` label when you create your PR, adding it after the fact as has no effect and we're forced to merge with red CI.

romancardenas, 2023-10-03:
As mentioned in [this PR](https://github.com/rust-embedded/riscv-rt/pull/124), perhaps you can fix this by isolating the Changelog job in a different workflow and configuring it to re-run whenever you add/remove labels to a PR.

PR #837 (2023-10-03): allow gha to detect label changes
Closes #775
closed 2023-10-03 by jessebraham