esp-rs/esp-hal #776: Provide a way to enable Peripherals without direct access to PeripheralClockControl

AnthonyGrondin, 2023-09-01:
Most peripherals require mutable access to PeripheralClockControl to be enabled, but there are some use cases, where we cannot have a direct access to it.

A use case would be to implement hardware acceleration for [esp-mbedtls](https://github.com/esp-rs/esp-mbedtls) we have to rewrite `extern "C"` bindings, while keeping the original signature. This means we cannot access the PeripheralClockControl from this context, to enable hardware peripherals.

PeripheralClockControl seems to be provided in the system struct, and then mutably borrowed back to enable peripherals. If it's not used for other cases, should we refactor so that peripherals gets enabled with their clock when instantiated, or enabled with a .enable(self) function?

MabezDev, 2023-09-04:
imo, long term I think we might actually want to remove the PeripheralClockControl from the public interface, and instead let the drivers manage the peripheral clocks internally. For now, though because PeripheralClockControl is a ZST you can just use some unsafe and pass initialized memory as the argument.

AnthonyGrondin, 2023-09-04:
Hi!

Do you have an example of how to do so?

In many modules, mbedtls creates a context out of thin air and then passes it to every function.

the `unsafe extern "C" fn *_init(context) {}` is expected to load the hardware acceleration module, and store its pointer into the context. 

So inside an `unsafe extern "C" fn`, I have to get the address of the peripheral, the peripheral clock control, initialize the peripheral and then store its pointer in the context to be used inside the other functions. 

I can ensure safety, on the peripheral at least, by making mbedtls take the peripheral, to make sure nobody else is using it.

bjoernQ, 2023-09-05:
I think Scott is thinking of something like this
```rust
    let mut pcc = unsafe {
        let pcc: MaybeUninit<PeripheralClockControl> = MaybeUninit::zeroed();
        pcc.assume_init()
    };
    pcc.enable(hal::system::Peripheral::Sha);
```

It works because PeripheralClockControl is zero-sized.

PR #817 (2023-09-26): No longer publicly expose the `PeripheralClockControl` struct
I have been meaning to do this for some time now, finally got around to it.

Rather than exposing the `PeripheralClockControl` struck via the `SystemParts` struct, the drivers just handle this all internally. Since `PeripheralClockControl` was already a ZST, I've just removed the `&mut self` argument of the `enable` function and instead converted it to an associated function. This type's scope was also limited to the crate-level.

This is obviously a pretty breaking change, but it's more than worth it IMO.

These changes were pretty [... cut]
closed 2023-09-26 by jessebraham