esp-rs/esp-hal #778: no_std issue with esp32c6-hal

blue-freedom-technologies, 2023-09-03:
We have no issues creating and flashing  std rust applications.

But when we create a no_std it does not work.

As an example, we have create a hello-world based in the [esp-template](https://github.com/esp-rs/esp-template), and it becomes stuck indefinitely.

[image]

Any advise would be welcome.

Cheers,

JurajSadel, 2023-09-04:
Hello @blue-freedom-technologies! I just tried creating a new C6 project and it works fine for me. Could you please provide us with more information like what exact options you use or if you somehow modified the template code?

blue-freedom-technologies, 2023-09-04:
Thank you @JurajSadel for your response. 

We have create a public [repo](https://github.com/blue-freedom-technologies/bug) with the generated code.

We all have ubuntu(s):
[image]

Our Rust environment:
[image]

Probably we should upgrade the esp32-c6 firmware ?

Cheers,

bjoernQ, 2023-09-04:
I guess you are using the USB connector on the right of the dev-board?

In that case it's using serial-jtag and to see the output you would need to choose it for esp-println/esp-backtrace like this
```
esp-backtrace = { version = "0.7.0", features = ["esp32c6", "panic-handler", "exception-handler", "print-jtag-serial"] }
esp-println       = { version = "0.5.0", default-features=false, features = ["esp32c6","jtag_serial"] }
```

Or you can just use the other USB port

JurajSadel, 2023-09-04:
I tried to run your [linked repo](https://github.com/blue-freedom-technologies/bug) on Ubuntu and it works fine for me. Could you please double-check the connection as @bjoernQ suggested?

blue-freedom-technologies, 2023-09-04:
Hi @bjoernQ , @JurajSadel 

Yep, we have try that also..

But the result is:
[image]

JurajSadel, 2023-09-04:
Do you use "default" Ubuntu or is it the WSL subsystem?

blue-freedom-technologies, 2023-09-04:
We all use real Ubuntu(s) no WSL windows subsystems.

JurajSadel, 2023-09-04:
If you list your USB ports, what does the C6 board look like? `/dev/ttyXXXX`

JurajSadel, 2023-09-04:
Glad you figured it out! Can this issue be closed now?

blue-freedom-technologies, 2023-09-04:
@JurajSadel  Yep, we have closed the issue.

closed 2023-09-04 by blue-freedom-technologies