esp-rs/esp-hal #788: Decide on MSRV

bjoernQ, 2023-09-13:
See https://github.com/esp-rs/esp-hal/pull/787

I like the `CURRENT_VERSION - 1` idea and with https://github.com/esp-rs/esp-hal/pull/779/commits/7b45dca3c088183c9e6da86c362c4fb068ff8954 that should be easy to implement

jessebraham, 2023-09-13:
Is there any reason not to just use `1.67.0`, which is currently required for the async example, and remove the distinction there altogether? I don't think there are really any new language features we want to take advantage of (correct me if I'm wrong) so giving some headroom would be nice.

bjoernQ, 2023-09-14:
Also fine for me. Don't have a strong opinion on this other than having to pin a (transitive) dependency because of the MSRV is not a good thing

MabezDev, 2023-09-14:
I don't think we should blindly upgrade to `CURRENT_VERSION - 1` whenever its available, but if there is a feature we want we can just go ahead and update provided its at least one version behind from current stable. We could also of course scrap that idea and just address it on a case-by-case basis.

bjoernQ, 2023-09-14:
#779 additionally requires `toml_edit` to be pinned in `esp32c6-lp-procmacros` 

If we need to bump MSRV we should make sure it's at least `CURRENT_VERSION - 1` or older. 
But we won't bump the MSRV just because we can

PR #798 (2023-09-19): Bump the MSRV to `1.67`, check the `defmt` feature in MSRV check
- Bumps the MSRV of each package to `1.67`
- Check the `defmt` feature in the MSRV check
- Add the `esp32c6-lp-hal-procmacros` package to the VS Code workspace

This unifies the various MSRV differences between RISC-V/Xtensa/async/defmt, so I think it's a nice improvement. Much simpler now, just a single version (as it should be).

Not sure why defmt was stated as requiring the latest `stable` release previously, but it seems to build just fine in our MSRV check. If there is more to this, please provide some documentation so I ca [... cut]
closed 2023-09-19 by jessebraham