esp-rs/esp-hal #791: Is there a reason for not exposing a Write Only SPI Instance with CS (which implements Fullduplex)?

VincentWo, 2023-09-14:
Currently it does not seem to be possible to create a SPI Device that uses SCK + CS + MOSI, but this setup can be quite useful for "output only" devices like displays, currently one has to either use `new_no_cs_no_miso` and implement the CS in software or use the half-duplex constructor, which decreases usability (and there does not seem to be a way to use async DMA with the half-duplex constructor AFAIK)

Is there a reason there does not exist a `new_no_miso` constructor? Because if it would be possible to add such a constructor (or maybe unify it with the `new_no_cs_no_miso` constructor by making the CS pin optional) it would be great!

MabezDev, 2023-09-14:
No reason, I think it was just missed. Would you like to PR adding this new constructor?

VincentWo, 2023-09-18:
Thanks for the quick response, it looks like somebody was faster than I :)

bjoernQ, 2023-09-18:
I guess we can close this. Thanks

closed 2023-09-18 by bjoernQ