esp-rs/esp-hal #800: Evaluate automating the releases publish

SergioGasquez, 2023-09-20:
This is done manually at the moment, and it might be worth to investigate adding a release workflow that automates it.

bjoernQ, 2023-09-21:
Just stumbled upon this: https://github.com/mindsers/changelog-reader-action

I haven't tried it and not sure if it helps - just leaving it here in case it might be useful

jessebraham, 2024-02-29:
I think I'm going to close this for the time being, releasing isn't too big of a deal and I don't think it's worth spending any time on this right now.

Maybe in the future we can revisit this.

jessebraham, 2024-03-07:
I've actually had a change of heart here, and think we should revisit this. With the addition of our self-hosted documentation, there are a number of things to consider for a release, and we should automate this as much as possible.

We can discuss this in one of our team meetings in the coming weeks, there is no immediate need.

jessebraham, 2024-03-11:
I will continue to add items as I think of them, but off the top of my head the general release process is as follows:

- Publish new packages
- Create a new release + tag
- Manually invoke the documentation workflow

I think this most of this process can probably be implemented in a single release workflow, at least the second and third points.

jessebraham, 2024-03-20:
I am looking into this on the side, between other projects. Will have something soon-ish probably.

MabezDev, 2024-06-05:
Given the complexity of our release process, especially with all the new crates, automation seems unfeasible at this time. Closing for now.

closed 2024-06-05 by MabezDev