esp-rs/esp-hal #803: Remove direct-boot support?

MabezDev, 2023-09-20:
With probe-rs now supporting the IDF bootloader format, and that it's impossible to use esp-wifi with direct boot mode I'm wondering if we should remove direct-boot support?

Personally, I'd rather focus our efforts on improving the main linker scripts than trying to maintain both, but if there are some compelling reasons to keep the direct boot linker scripts I'd like to hear them!

AnthonyGrondin, 2023-09-20:
My :two: :coin: 

I think eventually, direct-boot should be supported as a feature, since it's part of the functionalities this chip offers. I've personally never used it, but it has its advantages, and purpose (Like faster flash times, which can be considerable in repeated flashing operations).

As the `HAL` is being built, I don't think it's a hard requirement to support it, as there are changes done quite often. I've seen it break quite a few times over the pas year. I believe it could be looked at later, further down the line, once everything has stabilized, and more energy can be fully focused on it. 

These SoCs also support secure boot, so that's maybe a thing that could be looked at, at the same time.

bugadani, 2023-09-20:
I like having options, but if the only argument in favour of direct boot is flashing time, then my take is that the tooling should be improved, and the maintenance burden imposed by direct-boot dropped.

The ideal state of the world would be a single image format that can be either direct booted or loaded by the bootloader, but that's just wishful thinking on my part, it looks like :)

lucasvr, 2023-09-26:
Hi folks. Is there a technical reason why Wi-Fi won't work with direct-boot? Does it need special information from the metadata sections that are only declared by mcuboot, or is this a limitation of the current implementation of IDF?

bjoernQ, 2023-09-27:
> Hi folks. Is there a technical reason why Wi-Fi won't work with direct-boot? Does it need special information from the metadata sections that are only declared by mcuboot, or is this a limitation of the current implementation of IDF?

Not sure if there is really no way to get it working - back then we had a few general problems with direct-boot IIRC and we never really tried it again. Given that we might remove support for direct-boot it's probably a waste of time to look into that again. If we decide to keep direct-boot we should look into it again

MabezDev, 2023-11-06:
Based on the effort to maintain them, and the current state of scripts we have decided we will remove the direct boot linker scripts.

## Will it ever return?

We're not opposed to it returning, but the maintainability aspect must be addressed for us to consider it.

## Is there a way to use it even after the removal?

Yes, instead of linking to `-Tlinkall.x` (the built-in esp-hal linker scripts), you can maintain your own direct-boot linker scripts. By doing this you are at the mercy of changes in esp-hal needing linker script modifications to be reflected in your own scripts.

PR here: https://github.com/esp-rs/esp-hal/pull/903

PR #903 (2023-11-07): Remove direct boot support from esp-hal
Goodnight, sweet prince.

### Must

- [x] The code compiles without `errors` or `warnings`.
- [x] All examples work.
- [x] `cargo fmt` was run.
- [x] Your changes were added to the `CHANGELOG.md` in the proper section.
- [x] You updated existing examples or added examples (if applicable).
- [x] Added examples are checked in CI

### Nice to have

- [x] You add a description of your work to this PR.
- [x] You added proper docs for your newly added features and code.

Closes #803
closed 2023-11-07 by MabezDev