esp-rs/esp-hal #806: Add support for MCUboot

jessebraham, 2023-09-21:
MCUboot support was added for the ESP32-C3 in #49, we should continue adding support for additional devices. There is interest in this from other teams within Espressif currently, and as such it is likely a useful feature for some community members as well.

See [`esp32c3_hal/src/lib.rs`](https://github.com/esp-rs/esp-hal/blob/main/esp32c3-hal/src/lib.rs) as a reference. Note that additional changes to linker scripts are required, too.

- [ ] ESP32
- [ ] ESP32-C2
- [x] ESP32-C3
- [ ] ESP32-C6
- [ ] ESP32-H2
- [ ] ESP32-S2
- [ ] ESP32-S3

bjoernQ, 2023-11-16:
In the meantime, support for ESP32-C3 was removed

jessebraham, 2024-01-02:
I'm going to close this for now, as we have no plans on working on this any time soon. We can open a new issue if we decide to re-visit this.

closed 2024-01-02 by jessebraham