esp-rs/esp-hal #818: CI is not able to catch breaking errors

bugadani, 2023-09-26:
In the past week, the following build errors have been introduced with green CI:

 - This line shouldn't compile with `defmt` enabled: https://github.com/MabezDev/esp-hal/blob/b910e871ef9d1661c9a81104ca18be24a56573dc/esp-hal-common/src/lib.rs#L166
 - https://github.com/esp-rs/esp-hal/pull/812/files#r1336619682 `embassy-time-systick` and `async` are now incompatible

I can't explain the first one. I can compile examples locally even with defmt enabled and that shouldn't currently be possible - my pet project breaks "as expected".

For the second one, it looks like `cargo check` is insufficient to detect some errors.

bugadani, 2023-09-26:
I guess technically this will be helped by #809 ...

jessebraham, 2023-09-26:
Just a couple random comments for background:

- We used `cargo check` before because the only linker errors that really happened previously were due to the image formats, and we had explicit `cargo build` checks for these. I think it might be time to just use `cargo build` everywhere now, though, given that there are plenty of other sources of linker errors.
- CI has long been a thorn in my side, and we have re-written it a couple times now. Maybe we need to spend some time thinking about how we can improve this process and make it less prone to human error and bitrot

jessebraham, 2024-02-28:
We have completely reworked our CI workflow at this point, and while there are still some lingering issues I think most/all of them have their own issues. So, I hope this has been mostly resolved in the meantime, and I'm going to go ahead and close it.

As usual, if you feel this is the wrong thing to do then please feel free to re-open this issue or create a new one.

closed 2024-02-28 by jessebraham