esp-rs/esp-hal #842: Allow creating multiple `SHA` instances and / or resume state.

AnthonyGrondin, 2023-10-10:
While implementing SHA hardware acceleration in https://github.com/esp-rs/esp-mbedtls/pull/19, we've struck on a tricky edge-case, where in a real utilization scenario, the MbedTLS library needs to be able to quickly switch between different SHA operations, and also to be able to clone a context, and resume work for it.

bjoernQ, 2023-10-11:
Probably we should make this possible for all the crypto accelerators

AnthonyGrondin, 2023-10-16:
Hi!

I'm currently blocked on this for https://github.com/esp-rs/esp-mbedtls/pull/19 but I don't really know what would be the best way to do it. If anyone wants to take care of it, I can then implement hardware accelerated SHA1, SHA256 and SHA512 for `esp-mbedtls`.

I haven't looked currently into `AES` and the other crypto accelerators, but I assume it would be the same thing.

bjoernQ, 2023-10-17:
I hope I can take a swing on this but won't be able to get to this too soon. So, if someone else wants to work on this feel free to do so

My guess is that we need to save the current state after any operation and restore it before continuing an in-progress operation (see https://github.com/espressif/esp-idf/blob/8fc8f3f47997aadba21facabc66004c1d22de181/components/mbedtls/port/sha/block/sha.c#L19-L27 )

Will require API changes

closed 2024-08-19 by MabezDev