esp-rs/esp-hal #855: Wrong timing in `smart-led-adapter`

bjoernQ, 2023-10-17:
At least on ESP32 and ESP32-S2 the timing of the signals seems wrong:
- https://github.com/esp-rs/esp-hal/blob/0c8dd59163472133870a62b6248f50953f02defc/esp32-hal/examples/hello_rgb.rs#L35
- https://github.com/esp-rs/esp-hal/blob/0c8dd59163472133870a62b6248f50953f02defc/esp-hal-smartled/src/lib.rs#L39

Apparently, some LEDs are much, much more tolerant to incorrect timing - so it's often working this way. But e.g. on esp-buddy board the LED is very strict about timings

We should definitely validate timings with a logic analyzer when fixing it, this time.

jessebraham, 2023-10-17:
Related to #74 I guess

bjoernQ, 2023-10-17:
> Related to #74 I guess

Oh, thanks for linking that issue! Yes, this is really a bad inconsistency - totally forgot about that issue

bjoernQ, 2023-11-08:
I double checked and with all my WS2812 LEDs it works as it is - so there seem to be something interesting going on with ESP-BUDDY

bjoernQ, 2023-11-08:
Closing (for now) since it seems to be something special about one special (now unavailable) dev-board

closed 2023-11-08 by bjoernQ