esp-rs/esp-hal #863: example can't compile because esp32s-hal-common isn't the latest

maayanhaimdulberg, 2023-10-22:
example (and probably not only this one) https://github.com/esp-rs/esp-hal/blob/main/esp32s3-hal/examples/i2c_bmp180_calibration_data.rs can't compile.

I2C::new() introduces a new argument (PeripheralClockControl) in the newest esp32s-hal-common, but the newest example version isn't pointing to the latest one.

[image]

bugadani, 2023-10-22:
> I2C::new() introduces a new argument (PeripheralClockControl) in the newest esp32s-hal-common

This isn't correct, the argument was instead removed recently.

bjoernQ, 2023-10-23:
Our main branch is not the currently released version but the "next" version. The next release will contain a couple of breaking changes so examples in `main` won't currently work (see above)

(Closing this - feel free to re-open if you feel the issue isn't resolved)

closed 2023-10-23 by bjoernQ